micropython/micropython #5471: OTA: Error setting boot partition for ESP32

chtinnes, 2019-12-28:
I just wanted to use the ESP32 Partition class for OTA.
Everything seems to work fine, except setting a partition as boot partition does not work.

```
>>> from esp32 import Partition
>>> running = Partition(Partition.RUNNING)
>>> next = running.get_next_update()
>>> next.info()
(0, 18, 1441792, 1376256, 'ota_2', False)
>>> running.info()
(0, 17, 65536, 1376256, 'ota_1', False)
>>> next.set_boot()
I (57130) esp_image: segment 0: paddr=0x00160020 vaddr=0x3fff0018 size=0x00004 (     4) 
I (57140) esp_image: segment 1: paddr=0x0016002c vaddr=0x3fff001c size=0x012d0 (  4816) 
I (57140) esp_image: segment 2: paddr=0x00161304 vaddr=0x40078000 size=0x025b8 (  9656) 
I (57160) esp_image: segment 3: paddr=0x001638c4 vaddr=0x40080400 size=0x01848 (  6216) 
>>> boot = Partition(Partition.BOOT)
>>> boot.info()
(0, 17, 65536, 1376256, 'ota_1', False)
```
I inserted a few more logs to the set_boot() method:

```
>>> from esp32 import Partition
>>> running = Partition(Partition.RUNNING)
>>> running.info()
(0, 17, 65536, 1376256, 'ota_1', False)
>>> running.set_boot()
I (43650) Partition Module: Esp Partition type (0)!
I (43650) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3da1c (252444) map
I (43800) esp_image: segment 1: paddr=0x0004da44 vaddr=0x3ffb0000 size=0x025cc (  9676) 
I (43800) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xcc278 (836216) map
I (44270) esp_image: segment 3: paddr=0x0011c298 vaddr=0x3ffb25cc size=0x00b6c (  2924) 
I (44270) esp_image: segment 4: paddr=0x0011ce0c vaddr=0x40080000 size=0x00400 (  1024) 
I (44280) esp_image: segment 5: paddr=0x0011d214 vaddr=0x40080400 size=0x1263c ( 75324) 
I (44330) esp_image: segment 6: paddr=0x0012f858 vaddr=0x400c0000 size=0x00064 (   100) 
I (44330) esp_image: segment 7: paddr=0x0012f8c4 vaddr=0x50000000 size=0x00808 (  2056)
E (44340) Partition Module: esp_ota_set_boot_partition failed (ESP_ERR_INVALID_ARG)!
```
The boot partition is not set, because of *ESP_ERR_INVALID_ARG*. According to the IDF reference *partition argument was NULL or didn’t point to a valid OTA partition of type “app”*.
I did a check immediately before the esp_ota_set_boot_partition() method has been called and it turns out that the partition object is correct and partition type is correct as well.

Has anyone successfully tested set_boot()? Might this be a problem of the IDF version (i am using 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df)?

saviourxx, 2020-01-15:
Mq_Rx. 62 : AA 00 01 00 33 00 01 00 33 30 FE 00 00 01 02 03 00 28 68 74 74 70 3A 2F 2F 31 39 32 2E 31 36 38 2E 31 2E 33 30 3A 38 30 37 30 2F 61 70 70 6C 69 63 61 74 69 6F 6E 2E 62 69 6E 93 1B D4 BB 
mqtt rx ota process!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
otaver:  b'\x01\x02'
otaaddrlen:  40
otaaddr:  b'http://192.168.1.WiFi network! Attempting to connect to server...[0m
[0;32mI (85534) esp_https_ota: Starting OTA...[0m
[0;32mI (85534) esp_https_ota: Writing to partition subtype 16 at offset 0x10000[0m
[0;32mI (101854) esp_https_ota: Connection closed, all data received[0m
[0;32mI (101864) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (101864) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x47300 (291584) map[0m
[0;32mI (102034) esp_image: segment 1: paddr=0x00057328 vaddr=0x3ffbdb60 size=0x036bc ( 14012) [0m
[0;32mI (102044) esp_image: segment 2: paddr=0x0005a9ec vaddr=0x40080000 size=0x00400 (  1024) [0m
[0;32mI (102044) esp_image: segment 3: paddr=0x0005adf4 vaddr=0x40080400 size=0x0521c ( 21020) [0m
[0;32mI (102064) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xf8ae0 (1018592) map[0m
[0;32mI (102644) esp_image: segment 5: paddr=0x00158b00 vaddr=0x4008561c size=0x1436c ( 82796) [0m
[0;32mI (102694) esp_image: segment 6: paddr=0x0016ce74 vaddr=0x400c0000 size=0x00064 (   100) [0m
[0;32mI (102694) esp_image: segment 7: paddr=0x0016cee0 vaddr=0x50000200 size=0x00808 (  2056) [0m
[0;32mI (1020 vaddr=0x320 size=0x47300 (291584) map[0m
[0;32mI (102884) esp_image: segment 1: paddr=0x00057328 vaddr=0x3ffbdb60 size=0x036bc ( 14012) [0m
[0;32mI (102884) esp_image: segment 2: paddr=0x0005a9ec vaddr=0x40080000 size=0x00400 (  1024) [0m
[0;32mI (102894) esp_image: segment 3: paddr=0x0005adf4 vaddr=0x40080400 size=0x0521c ( 21020) [0m
[0;32mI (102914) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xf8ae0 (1018592) map[0m
[0;32mI (103484) esp_image: segment 5: paddr=0x00158b00 vaddr=0x4008561c size=0x1436c ( 82796) [0m
[0;32mI (103534) esp_image: segment 6: paddr=0x0016ce74 vaddr=0x400c0000 size=0x00064 (   100) [0m
[0;32mI (103534) esp_image: segment 7: paddr=0x0016cee0 vaddr=0x50000200 size=0x00808 (  2056) [0m
[0;32mI (103604) simple_ota_example: Firmware upgrade download success[0m
ota success!
Mq_Tx. 17 : AA 00 05 00 06 00 01 00 06 03 FE 04 00 2B 8B 77 BB 
I (103644) wifi: state: run -> init (0)
I (103644) wifi: pm stop, total sleep time: 64821945 us / 86945537 us

I (103644) wifi: new:<12,0>, old:<12,0>, ap:<255,255>, sta:<12,0>, prof:1
[0;32mI (103654) wifi: STA_DISCONNECTED, reason:8[0m
E (103654) wifi: esp_wifi_connect 1145 wifi not start
I (1036640x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

I kept track of your logs and did many trys. 
I think that the problem is the partition table and  partition in flash.
I succeed when I try:
1.build this repository. https://github.com/thingsboard/esp32-ota
2.flash bootloader at 0x1000, partitions at 0x8000, ota data at 0xd000
3.flash you app code in 0x10000
4.finally, try the command in you log. (from esp32 import Partition..)
All things will be right and thanks for you log.

chtinnes, 2020-01-15:
Hi,

I am not sure, why i have to use the code provided at thingsboard/esp32-ota.
I thought it should be possible with ESP32 Micropython port. In case it is related to my partition table, here it come:

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,   0x4000,
otadata,  data, ota,     0xd000,   0x2000,
phy_init, data, phy,     0xf000,   0x1000,
ota_1,    app,  ota_1,   0x10000,  0x150000,
ota_2,    app,  ota_2,   0x160000, 0x150000,
vfs,      data, fat,     0x2B0000, 0x150000,

I just get a binary that i flash to the device via:

esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 firmware-esp32.bin

So i am using 0x1000 as start addr which should be okay. 
Actually, loading the partitions seems to work fine but just the setting ot the boot partition does not work.

saviourxx, 2020-01-16:
Hi Chtinnes,

This is a simple example at thingsboard/esp32-ota that the log can show you more about the partition information by default.
We don't need to consider what app(Micropython/FreeRTOS/while(1)...) in factory or ota partitions, but the others are very import for ota.
So you can flash the partitions in the right way,  burn app.bin in factory part and two ota parts.
Then run these commands will be right.
next = running.get_next_update() 
next.set_boot()
boot = Partition(Partition.BOOT)

I can't use esptool.py in my wsl2 system in win10, so i choose flash_download_tools instead of esptool. https://www.espressif.com/sites/default/files/tools/flash_download_tools_v3.6.8_0.zip
It's necessary to erase the flash before burn anything.

And here is my partition table with 4MB flash:
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
ota_0,    app,  ota_0,   0x10000, 0x1A0000,
ota_1,    0,    ota_1,   0x1B0000,0x1A0000,
vfs,      data, fat,     0x350000,0xB0000,

mrwhy-orig, 2020-01-31:
I jsut finished my tests, an they are working fine. I've written a short description of what i did in the forum.
https://forum.micropython.org/viewtopic.php?f=18&t=7066#p43583

chtinnes, 2020-04-07:
The problem was my partition table. 
Instead of using ota_0 for first and ota_1 for second slot I used:

ota_1,    app,  ota_1,   0x10000,  0x150000,
ota_2,    app,  ota_2,   0x160000, 0x150000,

This seems to lead to the ESP_ERR_INVALID_ARG error.

closed 2020-04-07 by chtinnes