micropython/micropython #5480: Serial repl freezes after initializing 16 gpio pins as output in esp32.

gauravgupta45, 2020-01-01:
I am working on a 4x4x4 led cube which uses 20 pins(16+4) as an output. 
I am using the following command to initialize 16 pins as output(to use led's on it)

```python
from machine import Pin

column = [0,15,2,4,16,17,5,18,19,21,3,1,22,23,13,12]   #These are the GPIO pins of ESP32
colpin={}
for i in range(0,16):
    colpin[i]=Pin(column[i],Pin.OUT)
```
After running it in ESP32, my REPL freezes. I'm using Putty for Serial Comm.

But the above piece of code works fine if I decrease the range from 16 to 10.

```python
from machine import Pin

column = [0,15,2,4,16,17,5,18,19,21,3,1,22,23,13,12]
colpin={}
for i in range(0,10):
    colpin[i]=Pin(column[i],Pin.OUT)
```

robert-hh, 2020-01-01:
GPIO1 and GPIO3 are the ports used by UART0, which in turn is used for REPL. So it's no surprise that REPL stops if you take away the UART from it.

gauravgupta45, 2020-01-01:
Got it. Thanks!

closed 2020-01-01 by gauravgupta45