micropython/micropython #5481: Aliasing ubinsacii to binsacii

osresearch, 2020-01-01:
In [this comment](https://github.com/micropython/micropython/issues/4052#issuecomment-413509146) @robert-hh said:
> In Micropython, the module is called ubinascii, with binascii as alias.

My [efr32 port](https://github.com/osresearch/micropython/tree/efm32/ports/efm32) doesn't have this alias, although I would like for it to rename or alias both `ubinascii` as well as `uos` so that scripts using `import os` don't have to be specialized for Micropython.  How should these aliases be created?

robert-hh, 2020-01-01:
By this commit: https://github.com/micropython/micropython/commit/d2384efa809953152c57cbda4c339dfbaa64cf29
automatic weak links were introduces. So if you import binasciii and it is not present, it will find ubinascii. So it depends, which version the efr32 port is based on. I recall that it was later.

osresearch, 2020-01-01:
Ah, I had to define `MICROPY_MODULE_WEAK_LINKS` in the `mpconfigport.h` for the aliases to show up.  Thanks!

closed 2020-01-01 by osresearch