micropython/micropython #5484: More than ten characters, print func cannot print

SummerLife, 2020-01-02:
Hello everyone,

I encountered some problems when migrating MicroPython 1.12 version to RTOS (RT-Thread), 
[https://github.com/RT-Thread-packages/micropython/pull/150](url)

and now the problem is that if there are more than ten characters, print cannot output a string, As shown below：

[image]

This problem has not been encountered in previous migrations, such as before 1.10.
I have spent a day studying this issue, but with no results.
I probably know that after compiling, the length of the print string object is wrong and becomes 0, so this problem is caused.

Because I upgraded all the files in the py directory, I think this problem may be caused by a configuration error. I hope someone familiar with version 1.12 can give me some tips to solve the problem, thank you very much.

dpgeorge, 2020-01-06:
This looks quite strange and I can't think of a reason for it.  You'll need to do some debugging to find out exactly what is going wrong.

SummerLife, 2020-01-06:
> This looks quite strange and I can't think of a reason for it. You'll need to do some debugging to find out exactly what is going wrong.

Thank you very much for your reply.

If I use the os.listdir () command to print the file information in the current folder, the situation is similar, this is probably the case:

```
[ " ", " ", " ", " " ]
```

In other words, the file name will not be displayed, so I think the problem is because some types of string objects cannot be printed. I will do more debugging.

SummerLife, 2020-02-20:
I have tried again and succeeded, thank you anyway!!!

closed 2020-02-20 by SummerLife