micropython/micropython #5487: MICROPY_PORT_DEINIT_FUNC called after gc_sweep_all

amirgon, 2020-01-03:
I'm using `MICROPY_PORT_DEINIT_FUNC` to release hardware resources upon soft reset.
Unfortunately, it's called only after gc ram is deallocated by `gc_sweep_all`.  
My port uses gc allocated ram, and when `MICROPY_PORT_DEINIT_FUNC` is called it's too late, gc ram was already deallocated.

https://github.com/micropython/micropython/blob/10709846f38f8f6519dee27694ce583926a00cb9/ports/esp32/main.c#L145-L153

Is there any particular reason `gc_sweep_all` is called before `mp_deinit`?
Is it reasonable to move `gc_sweep_all` after `mp_deinit`?

stinos, 2020-01-03:
Note `MICROPY_PORT_DEINIT_FUNC` is a MicroPython core implementation detail and is not really meant to be used for customisations, and is also not a good fit for it: it's a macro so if the esp32 port in the future decides to put it to use, your build will break and getting your current code to be called in the same way will require source file changes.

amirgon, 2020-01-03:
@stinos Could you suggest a different way to detect a soft reset?  
How else could a user driver release hardware resources upon soft reset?

stinos, 2020-01-03:
Sorry have zero experience with that, no idea if there are extension points for that. There should be I guess. Also from a symmetry point of view it could make sense to have `gc_sweep_all` after `mp_deinit` since `mp_init` is also after `gc_init`. Where do you initialize your driver btw?

amirgon, 2020-01-03:
> Where do you initialize your driver btw?

My driver has an `init` and `deinit` functions that have micropython API, so they can be called (and are called) from the micropython script.   
In case of soft reset, however, I need to call the `deinit` function implicitly, before `gc_sweep_all`.

Ideally it could be great if we had the option to register a "soft reset callback" on both C code and micropython code.

dpgeorge, 2020-01-06:
If you need to clean up particular instances of a class then you might consider implementing `__del__` on that class, which is guaranteed to be called by `gc_sweep_all`.  See eg how `esp32.RMT` works, calling `deinit` on all the live instances at soft reset.

If this doesn't work then either there needs to be a user hook (at the C level) that runs before `gc_sweep_all`, or some generic way of registering a Python function that is called at reset -- and `sys.atexit()` is probably good to use for that.

amirgon, 2020-01-06:
> `__del__` on that class, which is guaranteed to be called by `gc_sweep_all`

When `__del__` is called, what assumptions do I have on gc ram? Can I use memory allocated by gc? Currently my code crashes when I try to.

> some generic way of registering a Python function that is called at reset -- and `sys.atexit()` is probably good to use for that

Good idea. But it's not implemented for ESP32, only for the unix port.  
Would you like a PR for calling `sys.atexit()` on ESP32?

dpgeorge, 2020-01-07:
> When `__del__` is called, what assumptions do I have on gc ram? Can I use memory allocated by gc?

You should be able to access GC blocks, just not allocate or free them.  Not sure why it would be crashing.

> Would you like a PR for calling `sys.atexit()` on ESP32?

Yes that would be good.