micropython/micropython #5488: ESP32: Crash after pushing Ctrl-C

robert-hh, 2020-01-03:
I stumbled over a strange crash situation. It happens if i type the follwoing two lines on REPL
```
b=bytearray(8192)  # size does not matter, just has to be large
b
```
While b is dumped, push Ctrl-C. That will not stop the dump, but at the end I get:
```
 NLR jump failed, val=0x3ffc1d44
E (17543) wifi: esp_wifi_connect 1156 wifi not start
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5000
load:0x40078000,len:10404
load:0x40080400,len:5680
entry 0x400806bc
```
and the device reboots. The code is a fresh build. The device was erased before flashing.

MicroPython v1.12-35-g10709846f-dirty on 2020-01-03; ESP32 module with ESP32

mikewadsten, 2020-01-05:
Related: #1667, #1722, #1738, #4921

dpgeorge, 2020-01-12:
Also related #5469 (two ctrl-C in a row lead to esp32 crash).

dpgeorge, 2020-02-07:
Fixed by 5a91cd9ff3e97b74ad6c9a615f810b8fcebbe110

closed 2020-02-07 by dpgeorge