micropython/micropython #5492: ESP32 Static IP not being assigned despite using ifconfig()

ekendra-nz, 2020-01-04:
Using ESP32 Devkit flashed with MP Build 2 Jan 2020.

I'm trying to get a static IP from the router. I'm using the tuple param:

`wlan.ifconfig(("192.168.1.77", "255.255.255.0", "xxx.xxx.x.xx", "8.8.8.8"))
`
Connects to LAN but DHCP is still just assigning it an IP and ignoring the requested static IP 192.168.1.77

ekendra-nz, 2020-01-04:
Requested static IP 192.168.1.77 is not within DHCP range.

robert-hh, 2020-01-04:
The works for me on an ESP32 (and others). I call wlan.ifconfig() before wlan.connect().

ekendra-nz, 2020-01-05:
> The works for me on an ESP32 (and others). I call wlan.ifconfig() before wlan.connect().

Same.

> Take care to go through a hard reset, because on a soft reset the device uses the previous assigned IP address.

Did. I think I just have a rubbish modem/router. I can't even assign a static IP via MAC address with it. Mine came free with ISP sign-up. I may need to get something less frustrating to work with.

My Home Assistant setup on a Rasp Pi manages to get a fixed IP from the stupid thing. So it is capable, just the ipconfig() isn't talking to it the way it wants.

IveJ, 2020-01-05:
Hi Ekendra,

We can not see IP addr of router, how about your config of router, as
mention 192.168.1.77 not in range DHCP server.

Nice.

ekendra-nz, 2020-01-06:
Maybe I'm just doing it wrong?
  ```

import esp
import machine
import network
from secret import Wifi_deets
  

# connect to WIFI function
def wifi_connect():

    # --------- WIFI deets object -----------
    wifi_creds = Wifi_deets()

    wlan = network.WLAN(network.STA_IF) 

    if wlan.isconnected():
        print("Already connected to wifi.")
        return

    print('connecting to network...')
    wlan.active(True)
    wlan.ifconfig((wifi_creds.static_ip, wifi_creds.network_mask, wifi_creds.gateway_ip, wifi_creds.dns_server))
    wlan.connect(wifi_creds.essid, wifi_creds.password)
    wlan.config(dhcp_hostname=wifi_creds.device_name)

    while not wlan.isconnected():
        machine.idle() # save power while waiting
 
    deets = wlan.ifconfig() # get the interface's IP/netmask/gw/DNS addresses

    print("Network Config:")
    print("IP:", deets[0])
    print("Netmask:", deets[1])
    print("Gateway:", deets[2])
    print("DNS:", deets[3])
 
```

robert-hh, 2020-01-06:
Try to delete this line:
`wlan.config(dhcp_hostname=wifi_creds.device_name)`
It is not required for a static IP address.

ekendra-nz, 2020-01-06:
> Try to delete this line:
> `wlan.config(dhcp_hostname=wifi_creds.device_name)`
> It is not required for a static IP address.

Ok. I've done that. Same result. Interesting behaviour is that when I print the wlan.ipconfig() towards the end, it shows me the requested static IP address when, in fact, the router has assigned another one via DHCP.

robert-hh, 2020-01-06:
a) can you access your board with the static IP address? Or reach out from the board with that address?
b) can you reset the DHCP table in the router? It may have still a link between MAC and IP address.

zurgeg, 2020-03-25:
From my experience with ISP provided routers, you should have to go to 192.168.1.1 (or just whatever your router's admin page is) and find your devkit then give it a static IP address. Also, it would be helpful if you could tell me your router name so I could find the instructions for it and give you exact info on how to do it. Thanks!

ekendra-nz, 2020-05-14:
@zurgeg yes. What you say is true. I threw that modem in the bin and got one that allows for basic functions such as assigning a static IP. All good now although the 
[image]

Thanks everyone for their help.

closed 2020-05-14 by ekendra-nz