micropython/micropython #5495: NRF - Frozen Modules not Available in REPL

pacmac, 2020-01-06:
3 Files in freeze folder:

```
freeze % ls -all
total 88
drwxrwxrwx  1 pac  staff  16384  6 січ 14:42 .
drwxrwxrwx  1 pac  staff  16384  6 січ 14:42 ..
-rwxrwxrwx  1 pac  staff    118  5 січ 21:18 flashbdev.py
-rwxrwxrwx  1 pac  staff   1463  7 кві  2019 sh.py
-rwxrwxrwx  1 pac  staff     63  5 січ 21:18 test.py
```

Frozen Folder is called during make an built with MPY:

```
make -j8 BOARD=holyiot_18010 SD=s140 FROZEN_MPY_DIR=freeze MICROPY_FATFS=1
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Warning: FROZEN_MPY_DIR is deprecated in favour of FROZEN_MANIFEST
.....
mkdir -p build-holyiot_18010-s140/py/
MPY freeze/test.py
MPY freeze/sh.py
MPY freeze/flashbdev.py
Create build-holyiot_18010-s140/pins_gen.c
```

3 Frozen Modules not available in REPL:

```
>>> help('modules')
__main__          machine           test              uos
ble               math              time              ustruct
board             micropython       uarray            utime
builtins          random            ubluepy
gc                sys               ucollections
Plus any modules on the filesystem
```

pacmac, 2020-01-06:
Strange, I changed permissions on the 2 symlinked files from 777 to 666 and they now appear !

```
 help('modules')
__main__          gc                sh                ubluepy
ble               machine           sys               ucollections
board             math              test              uos
builtins          micropython       time              ustruct
flashbdev         random            uarray            utime
Plus any modules on the filesystem
>>> 
```

But it seems there are a lot of file system functions missing, I cannot even change directory or make a directory, which means I cannot transfer files using RSHELL :-(

```
import os
>>> dir(os)
['__class__', '__name__', 'remove', 'sep', 'dupterm', 'ilistdir', 'listdir', 'stat', 'uname']
```

mzdaniel, 2020-01-08:
NRF is in the process of gain VFS functionality. AFAICT, rshell works after applying https://github.com/micropython/micropython/pull/5472