micropython/micropython #5499: Enhancement: please add support for the ADR assembler opcode

IowaDave, 2020-01-06:
Please add support for the ADR opcode in micropython for microbit. We need to utilize a temporary data table inside an assembler def.

Using 'adr' gives an error something like: "unsupported Thumb command 'adr' "

Desired usage:

def example():
    b(LABEL1)
    label(LABEL2)
    data(4, 1, 2, 3, 4, 5, 6, 7, 8)
    label(LABEL1)
    adr(r1, LABEL2) # r1 =  address of where data starts
# or, this is how it appears in some ARM/Thumb documentation:
#  adr(r0,=LABEL2) # r1 = address of LABEL2
    ldr(r0, [r1, 0]) # return value pointed to by LABEL2

If adr opcode cannot be supported, then please suggest a workaround using available opcodes. Thank you!

dpgeorge, 2020-01-07:
Which microbit firmware do you use, what version is it (`import os; os.uname()`)?

A workaround would be to use the `data` directive, eg:
```python
@micropython.asm_thumb
def example():
    data(2, 0xa101) # adr r1, <data address>
    ldr(r0, [r1, 0]) # 0 can be 0, 4, 8, 12, 16 etc to index the data
    b(SKIP)
    data(4, 1, 2, 3, 4, 5, 6, 7, 8)
    label(SKIP)
```
This works but you must guess the offset of the data array, and it must be a forward reference.

IowaDave, 2020-01-07:
Thank you, I will study this workaround. Don't intuitively understand how data(2, 0xa101) puts an address into r1, but I'm a good learner. Will update this thread when I get it working, if you choose to leave it open.

Answering your question, os.uname() returned:

sysname='microbit', nodename='microbit', release='1.0.1', version='micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', machine='micro:bit with nRF51822'

Thank you. Very nice to receive a response from a team leader so soon. Appreciate what has been done to make python available for microbit.

dpgeorge, 2020-01-07:
> Don't intuitively understand how data(2, 0xa101) puts an address into r1, but I'm a good learner.

0xa101 is the machine code for `adr r1, +4`.  The formula is: `0xa000 | (rd << 8) | (offset / 4)`

>  os.uname() returned

So you're using official microbit firmware from https://github.com/bbcmicrobit/micropython , it's not actually generated from the code in this repository.

> Appreciate what has been done to make python available for microbit.

Great that you are enjoying using it!

IowaDave, 2020-01-07:
Ah, I see it now. Of course. The data() directive instructs the assembler to emit a machine language instruction which then gets executed at runtime. Makes sense. The workaround demonstrates a way to employ valid opcodes from outside the set that are recognized by the micropython parser. Very good. I will study the construction of the instruction. 

Many thanks! This resolves my issue, at least functionally. I would be happy to contribute effort toward writing a how-to on the assembler for microbit, if you think it would help the community.

IowaDave, 2020-01-07:
Two final words from me:

1. Tested the workaround, it works. 
2. Studied the calculation of the machine language instruction. See pages 197-198 of the [ARM Architecture Reference Manual in .pdf format](https://static.docs.arm.com/ddi0403/e/DDI0403E_B_armv7m_arm.pdf). Offset is from the program counter (PC) which is advanced by 8 words due to cpu instruction prefetch. The PC points to the b(SKIP) instruction. The data directive that follows it is offset by a further 4 words from the PC address. Calculating the instruction involves dividing the offset by 4 (i.e., offset >> 2), therefore the encoded offset is 1. Offsets have to be a multiple of 4 for this reason.

Question for you: page 198 of the document referenced states that offset can be up to a value of 1020 for "T1 Encoding", which it says includes "all versions of the Thumb instruction set." Noting that 1020 / 4 = 255, my question is this: will all values for the encoded offset -- 0x00 through 0xff, inclusive -- work here? Naturally I plan to find out.

This workaround is very empowering. Thank you very much! The following code demonstrates the example around a somewhat larger array of data words:

```python
from microbit import *

@micropython.asm_thumb
def example():
    data(2, 0xa101) #
    mov(r0, r0) # nop, preserves the offset
    b(SKIP)
    label(MYARRAY)
    data(4, 1, 2, 3, 4, 5, 6, 7, 8)
    data(4, 9, 10, 11, 12, 13, 14, 15, 16)
    data(4, 17, 18, 19, 20, 21, 22, 23, 24)
    data(4, 25, 26, 27, 28, 29, 30, 31, 32)
    data(4, 33, 34, 35, 36, 37, 38, 39, 40)
    data(4, 76, 54, 87, 65, 98, 76)
    align(2)
    label(SKIP)
    # point r1 to MYARRAY[40]
    # which contains the value 76
    add(r1, 160) # because 40 * 4 = 160
    ldr(r0, [r1, 0])
    
display.show(example())
```

dpgeorge, 2020-01-07:
> will all values for the encoded offset -- 0x00 through 0xff, inclusive -- work here?

Yes they should.

On nRF51822 (microbit MCU) it only supports T1 encodings (I'm pretty sure...) so if you want to try out other opcodes then just use T1.

peterhinch, 2020-01-07:
[PR1192](https://github.com/micropython/micropython/pull/1192) enabled data created with the `data` directive to be accessed via labels. Maybe that needs revisiting to see if someone can come up with a more elegant solution than my pseudo-instruction.