micropython/micropython #5500: Native modules don't compile on macOS for x64 / x86 arch

stepansnigirev, 2020-01-06:
I tried to compile the native module example (`examples/natmod/feature0`) on macOS and got the following error:

```
GEN build/features0.config.h
CC features0.c
LINK build/features0.o
Traceback (most recent call last):
  File "../../../tools/mpy_ld.py", line 964, in <module>
    main()
  File "../../../tools/mpy_ld.py", line 961, in main
    do_link(args)
  File "../../../tools/mpy_ld.py", line 937, in do_link
    load_object_file(env, file)
  File "../../../tools/mpy_ld.py", line 585, in load_object_file
    elf = elffile.ELFFile(f)
  File "/usr/local/lib/python3.7/site-packages/elftools/elf/elffile.py", line 67, in __init__
    self._identify_file()
  File "/usr/local/lib/python3.7/site-packages/elftools/elf/elffile.py", line 421, in _identify_file
    elf_assert(magic == b'\x7fELF', 'Magic number does not match')
  File "/usr/local/lib/python3.7/site-packages/elftools/common/utils.py", line 77, in elf_assert
    _assert_with_exception(cond, msg, ELFError)
  File "/usr/local/lib/python3.7/site-packages/elftools/common/utils.py", line 109, in _assert_with_exception
    raise exception_type(msg)
elftools.common.exceptions.ELFError: Magic number does not match
make: *** [build/features0.native.mpy] Error 1
```

On Ubuntu everything compiles. On Mac armv7m works, x86 and x64 - not.
I am using macOS Catalina v10.15.2.

dpgeorge, 2020-01-07:
> On Mac armv7m works, x86 and x64 - not.

I guess that the host linker on Mac (clang?) is producing an ELF that elftools does not recognise.  Try running `make V=1` to get full output and post it here.

stepansnigirev, 2020-01-07:
Here it is:
```
/bin/mkdir -p build
GEN build/features0.config.h
python3 ../../../tools/mpy_ld.py '-vvv' --arch x64 --preprocess -o build/features0.config.h features0.c
CC features0.c
gcc -I. -I../../.. -std=c99 -Os -Wall -Werror -DNDEBUG -DNO_QSTR -DMP_CONFIGFILE='<build/features0.config.h>' -fpic -fno-common -U _FORTIFY_SOURCE  -fno-stack-protector -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_DOUBLE  -o build/features0.o -c features0.c
LINK build/features0.o
python3 ../../../tools/mpy_ld.py '-vvv' --arch x64 --qstrs build/features0.config.h -o build/features0.native.mpy build/features0.o
qstr vals: factorial
qstr objs: 
Traceback (most recent call last):
  File "../../../tools/mpy_ld.py", line 964, in <module>
    main()
  File "../../../tools/mpy_ld.py", line 961, in main
    do_link(args)
  File "../../../tools/mpy_ld.py", line 937, in do_link
    load_object_file(env, file)
  File "../../../tools/mpy_ld.py", line 585, in load_object_file
    elf = elffile.ELFFile(f)
  File "/usr/local/lib/python3.7/site-packages/elftools/elf/elffile.py", line 67, in __init__
    self._identify_file()
  File "/usr/local/lib/python3.7/site-packages/elftools/elf/elffile.py", line 421, in _identify_file
    elf_assert(magic == b'\x7fELF', 'Magic number does not match')
  File "/usr/local/lib/python3.7/site-packages/elftools/common/utils.py", line 77, in elf_assert
    _assert_with_exception(cond, msg, ELFError)
  File "/usr/local/lib/python3.7/site-packages/elftools/common/utils.py", line 109, in _assert_with_exception
    raise exception_type(msg)
elftools.common.exceptions.ELFError: Magic number does not match
make: *** [build/features0.native.mpy] Error 1
```

dpgeorge, 2020-01-07:
So the problem is that gcc (=clang) is generating Mach-O files, but `mpy_ld.py` only supports ELF files.  Maybe `mpy_ld.py` will support Mach-O files one day, but for now a way to get it working would be to install a C cross compiler that can emit ELF files on Mac.

stepansnigirev, 2020-01-07:
I see. Thank you! I will keep using my linux machine for now then.

agatti, 2025-11-15:
> Does anyone have a tip for how to install a cross-compiler that can output ELF on Mac OS?

Assuming you want to target Apple M-family machines you can install homebrew's [aarch64-elf-gcc](https://formulae.brew.sh/formula/aarch64-elf-gcc) and [aarch64-elf-binutils](https://formulae.brew.sh/formula/aarch64-elf-binutils) packages.

> And would one actually be able to load such files as native modules on Mac OS?

You'd have to first replicate 1b92bda5b8e5e2db8e57c4b7134930e52bc5207a for AArch64, and #18366 should be merged in (or something like that, so it won't break the debug emitter even further).

Then I guess it's a matter of setting up the right defines and the usual build stuff.  In theory the Unix port's MPY file support could still work on macOS, `mp_unix_alloc_exec` looks like it should support macOS (I see mentions of the `__MACH__` define in there).  If not, that's where the fun part begins :)

For bonus points you could also add AArch64 support to the QEMU port and extend the Unix port's QEMU-based foreign arch tests to also run for AArch64.

Unfortunately I don't have such a machine, otherwise I might have given that a try.