micropython/micropython #5501: [nrf] issue when compiling st7789_mpy

memiks, 2020-01-07:
Hi all,

I tried to compile st7789_mpy driver for NRF5x but received error related to this line :
https://github.com/micropython/micropython/blob/54a2584de10103ba3e559de1034cc76962ec681e/ports/nrf/mphalport.h#L68

So i need to apply this patch :
-#define mp_hal_pin_write(p, v)   do { if (v) { mp_hal_pin_high(p); } else { mp_hal_pin_low(p); } } while (0)
+#define mp_hal_pin_write(p, v)   (v == 1) ? mp_hal_pin_high(p) : mp_hal_pin_low(p)

[mphalport.patch.txt](https://github.com/micropython/micropython/files/4028322/mphalport.patch.txt)

Please find logs attached:
[logs_mpypinetime_1_3.log](https://github.com/micropython/micropython/files/4028317/logs_mpypinetime_1_3.log)

Do you think it is relevant ?

BR,
Memiks.

dpgeorge, 2020-01-07:
> Do you think it is relevant ?

Yes that makes sense, to make `mp_hal_pin_write` act more like a function, rather than a statement.  Looks like this code was copied from the stm32 port which could also be changed in a similar way.

dpgeorge, 2020-01-14:
Thanks!  Squashed and merged in 4ab4bf3ec6e891a49a827b659d5d4764a40ee404

closed 2020-01-14 by dpgeorge