micropython/micropython #5508: stm32: pyb.CAN.init(extframe=False) excludes all frames

jimmo, 2020-01-09:
From the forum -- https://forum.micropython.org/viewtopic.php?f=2&t=7530  (I have not verified this myself though)

JohnieBraaf, 2020-01-09:
thnx @jimmo, see the details below: 

There apprears to be an issue in the CAN drivers which prevents receiving messages with non extended id's.

As a test I'm using a pyboard-D to send data between the 2 CAN ports.
This works fine when the can filter is set to extframe=True.
`can.init(mode=pyb.CAN.NORMAL,` extframe=True, prescaler=4, sjw=1, bs1=7, `bs2=1)`

However, when changed to False, there are no incoming frames detected.
I can see the frames with a CAN bus analyzer, but the rx callback does not trigger.
`can.init(mode=pyb.CAN.NORMAL,` extframe=False, prescaler=4, sjw=1, bs1=7, `bs2=1`

-----------------------------------------------

I think I might have found where this goes wrong.

I'm using CAN.MASK32 with a blank values like this:
`can.setfilter(bank=self.itf-1,` mode=pyb.CAN.MASK32, fifo=self.itf-1, params=(0x0, `0x0))`

Now when I trace back the code, I come to pyb_can.c, where the filtermasks are set.
They appear to be designed for extended identifier format judging from the high value 0x1FFFE000.

https://github.com/micropython/micropython/blob/bfbd94401d9cf658fc50b2e45896aba300a7af71/ports/stm32/pyb_can.c#L710-L713

Now when I change it to blank, I start receiving the packets.
`
filter.FilterIdHigh = 0x0000;
filter.FilterIdLow = 0x0000;
`
`
filter.FilterMaskIdHigh = 0x0000;
filter.FilterMaskIdLow  = 0x0000;
`

I don't understand the bitwise operators that are applied in the original code, maybe someone can have a look at it.

gooyle, 2020-10-15:
has it been fixed already?we met it again....

gooyle, 2020-10-25:
> d to hack the workaround in yourself or even better attempt a proper fix. The workaround works in any case
thx，a new version lib works，no preblems appear till now

JohnieBraaf, 2022-01-24:
While writing down the steps to reproduce and looking at the code I realized, I should have used MASK16 instead of MASK32.
Sorry for the confusion. I managed to receive non-extended id messages on non fd interfaces with the following filter:
can1.setfilter(bank=0, mode=pyb.CAN.MASK16, fifo=0, params=(0x0, 0x0, 0x0, 0x0))

JohnieBraaf, 2022-01-24:
@jimmo this issue can be closed

closed 2022-01-27 by dpgeorge