micropython/micropython #5515: Got "Malformed Mach-o file" error when make firmware on MacOS Catalina

pointcloudAI, 2020-01-10:
I want to build firmware on MacOS Canalina but got "Malformed Mach-o file" error.

My detail process as below :

I had set up ~/edp-idf and ~/esp/xtensa-esp32-elf

On ~/.bashrc :
`
export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH
export IDF_PATH=~/esp/esp-idf
`
Then go to ports/esp32 and perform "make V=1", but got "Malformed Mach-o file" error as below :
`
../../mpy-cross/mpy-cross -o build/frozen_mpy/flashbdev.mpy -s flashbdev.py  modules/flashbdev.py
make: ../../mpy-cross/mpy-cross: Malformed Mach-o file
make: *** [build/frozen_mpy/flashbdev.mpy] Error 1
`

jimmo, 2020-01-10:
Some discussion already on the forum -- https://forum.micropython.org/viewtopic.php?f=18&t=7571

dpgeorge, 2020-01-11:
I can't reproduce this problem.  On macOS Catalina 10.15.2 I get:
```
$ gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.12)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
And `mpy-cross` builds and executes fine out-of-the-box on latest master.

pointcloudAI, 2020-01-12:
Do you got "BFD: mpy-cross: unknown load command 0x32" when build may-cross?

`CC gccollect.c
LINK mpy-cross
BFD: mpy-cross: unknown load command 0x32
   text	   data	    bss	    dec	    hex	filename
 130999	  13512	    860	 145371	  237db	mpy-cross`

pointcloudAI, 2020-01-12:
@dpgeorge  would you please share your mpy-cross for me for testing ?
Please share a link for download or mail me : hanyu@pointcloud.ai .Thx

`./mpy-cross --help
dyld: malformed mach-o image: missing LC_DYSYMTAB
Abort trap: 6`

dpgeorge, 2020-01-12:
> Do you got "BFD: mpy-cross: unknown load command 0x32" when build may-cross?

No I didn't.  It builds fine without any error.

> would you please share your mpy-cross for me for testing ?

Sure.  You can get my version here: https://micropython.org/resources/mpy-cross-mac64

pointcloudAI, 2020-01-13:
> Sure. You can get my version here: https://micropython.org/resources/mpy-cross-mac64

Would you please share me below files for further investigation?
1) tools/mpy-tool.py
2) py/persistentcode.c
3) mpy-cross/mpy-cross.map

I will got below error even i modify the MPY_VERSION from 3 to 5 to align with your may-cross bin
`
  File "../../tools/mpy-tool.py", line 212, in extract_prelude
    while bytecode[ip] != 0xff:
IndexError: bytearray index out of range
`

jimmo, 2020-01-13:
Those files are in this repo, you can just browse to them on the master branch. It sounds like you're not building at the latest revision of the master branch (or at the v1.12 tag).

Could you run `git rev-parse HEAD`

pointcloudAI, 2020-01-15:
`git rev-parse HEAD`
7ef2f65114f092be6303c145a2560fdf522dcde0

dpgeorge, 2020-01-22:
@pointcloudAI you have the correct version of the code.  It sounds like the problem is with your build environment.