micropython/micropython #5523: Segmentation fault when MICROPY_PY_THREAD is enabled

tnogueira, 2020-01-12:
Starting from the [*embedding* example](https://github.com/micropython/micropython/tree/master/examples/embedding), and if i update `mpconfigport_minial.h` to enable `MICROPY_PY_THREAD` (`#define MICROPY_PY_THREAD           (1)`), the application exits with a *Segmentation fault* when running the `mp_init()` method. After some debugging, the segmentation fault seems to happen at [line 100 in `runtime.c`](https://github.com/micropython/micropython/blob/40057600b873b2a802de332d251492a762b32011/py/runtime.c#L100). 

I'm building using the unix port and running the application in a Debian machine.

Any idea of what could be causing the problem? 

Is there anything i'm missing if i want to enable multithreading support?

Thanks

jimmo, 2020-01-13:
You likely need to call `mp_thread_init()` before `mp_init()`.

It might be worth looking at one of the ports that uses threading (e.g. `ports/stm32/main.c` or `ports/unix/main.c`) and seeing the extra stuff they do if `MICROPY_PY_THREAD` is enabled.

tnogueira, 2020-01-13:
Thanks for the very quick reply. Indeed, i was missing a call to `mp_thread_init();`.
Problem solved.

closed 2020-01-13 by tnogueira