micropython/micropython #5538: boolean type cannot be detected anymore

stinos, 2020-01-15:
Commit d96cfd13e3a464862cecffb2718c6286b52c77b0 broke this:

```
mp_obj_t x = mp_const_true;
bool is_bool = mp_obj_is_type(x, &mp_type_bool);
```
is_bool is now false; as a workaround checking whether x equals mp_const_true or mp_const_false works, but I'm not sure what the best way is to get mp_obj_is_type working properly again.

stinos, 2020-01-15:
Note this check is also used in `mp_obj_is_integer` so some other things might be broken as well.

Jongy, 2020-01-18:
Hmm. Just like `mp_obj_is_type(x, &mp_type_int)` is not "complete" and one should be using `mp_obj_is_int`, I guess we should now have an `mp_obj_is_bool` helper for this purpose. From a quick search, `mp_obj_is_integer` and `arg_looks_integer` are the only direct users of `mp_obj_is_type` with `mp_obj_bool`. Others use `mp_obj_get_type` which handles booleans correctly.

`None` could also be affected but nothing uses `mp_type_NoneType` directly. Most uses of `mp_const_none` are direct comparison which works.

So either add such a helper, or update both call sites to use `mp_obj_get_type`. And a test for the incorrectly failing `mp_obj_is_integer` should be added as well.

stinos, 2020-01-19:
> Just like mp_obj_is_type(x, &mp_type_int) is not "complete"

Might be incomplete, but at least it does exactly what it is supposed to do. Good point about None, guess that has the exact same problem. And there are other 'incmplete' types like this (see obj.h, int/str/fun). So for consistency there should indeed be mp_obj_is_bool I guess but then the question is whether mp_obj_is_bool should work in 'strict mode' or also consider ints etc.

Still, it remains rather tricky a macro like mp_obj_is_type is part of the public API and the only thing guarding against incorrect use is a comment next to it, but I think it's too late to try and solve that now.

Jongy, 2020-01-19:
> Might be incomplete, but at least it does exactly what it is supposed to do

I meant that it doesn't work for all integers (i.e not for small ints)

Anyway, I have an idea based on some macros trickery that might solve this API problem by preventing misuses, as the docs of `mp_obj_is_type` explain. I'm off traveling the next few days, I'll post it aftwerwards.

dpgeorge, 2020-01-20:
Thanks @stinos for the report.  It's quite serious (in the sense that it breaks Python code, not just the C-level API) and I'm surprised that there are no tests to cover this case.  It's not that hard to fix, just need to see what the right way to do it is.

Here are some Python tests which now fail:
```python
'%c' % False
'{:d}'.format(False)
'{:f}'.format(False)
'{:20}'.format(False)
```
(and `False`->`True`).  Should also add a test to unix coverage tests to verify that `mp_obj_is_integer()` works as expected.

closed 2020-01-23 by dpgeorge

dpgeorge, 2020-01-24:
The problem is fixed by d9433d3e9497a4ec4cd51eb78d6ca602f310ce91 and tests added in 35e664d7790b123dab54faff766f41991f4620d1 and cb4472df42e6f34df2fe57f4f85786f6669171f7

commit d9433d3e (2020-01-23): py/obj.h: Add and use mp_obj_is_bool() helper.

Commit d96cfd13e3a464862cecffb2718c6286b52c77b0 introduced a regression in
testing for bool objects, that such objects were in some cases no longer
recognised and bools, eg when using mp_obj_is_type(o, &mp_type_bool), or
mp_obj_is_integer(o).

This commit fixes that problem by adding mp_obj_is_bool(o).  Builds with
MICROPY_OBJ_IMMEDIATE_OBJS enabled check if the object is any of the const
True or False objects.  Builds without it use the old method of ->type
checking, which compiles to smaller code (compared with the former
mentioned method).

Fi [... cut]