micropython/micropython #5539: esp32: cannot set RMT idle level

jonathanhogg, 2020-01-15:
Hi there,

Any chance of getting a hook to set the idle level for an RMT object? I need to be able to generate a precisely-timed active-low output, for which an RMT would be perfect except that it is hard-wired at the moment to idle low. It looks like it'd only be a small change in `esp32_rmt_make_new`.

Jonathan

MrSurly, 2020-01-15:
Per https://github.com/micropython/micropython/pull/5184#issuecomment-569462169, that feature is in the works.

jonathanhogg, 2020-01-15:
@MrSurly awesome! Any idea on a timeline?

I'm attempting to setup the build environment to generate my own firmware in the meantime…

MrSurly, 2020-01-15:
@mattytrentini is driving the RMT dev.  The branch is [here](https://github.com/mattytrentini/micropython/tree/esp32_rmt2).  I'm just watching the dev myself for possible future use.

jonathanhogg, 2020-01-15:
Ah! Apologies to you, @MrSurly, and to @mattytrentini for confusion.

Excitingly, and in my eagerness to get this working, I have managed to get a working build environment so I've produced firmware with an `idle_level` change incorporated…

MrSurly, 2020-01-15:
May I ask your use case?

jonathanhogg, 2020-01-15:
@MrSurly Sure!

I'm working on directly driving some drone-motor electronic speed controllers. They use a one-wire protocol ("DSHOT") that is very similar to NeoPixels, but operates at different speeds and with the line idling high.

Documentation on these things is incredibly thin-on-the-ground, so I'm working backwards from oscilloscope traces of the outputs of a working drone flight controller.

It's actually for a crazy art installation rather than anything like a quad copter, which is why I'm not just using a standard flight controller.

MrSurly, 2020-01-15:
Ah, okay, so a digital update to the old 1-2ms servo control.  My own interest is the RMT is also around reading and sending RC signals, though using PPM.

jonathanhogg, 2021-06-12:
Fix for this is in PR #6274, which will perhaps one day see mainline. In the meantime I am just using my own build so this issue isn't really doing anything useful.

closed 2021-06-12 by jonathanhogg