micropython/micropython #5541: esp32: build latest master (3032ae115) with BLE

petrkr, 2020-01-15:
Latest build seems to be broken (or maybe I do something wrong).

Have checked out esp-idf on v3.3.1 with --recurse
Have checked master micropython with --recurse too...

but getting 
`$esp-idf/components/nimble/esp-hci/src/esp_nimble_hci.c:23:29: fatal error: sysinit/sysinit.h: No such file or directory`

with some debugging there probably missing includes from `lib/lib/mynewt-nimble/...` with some added INC +=  was able move far, but without knowleadge it is like trial and error method.

Petr

jimmo, 2020-01-15:
This looks like your esp-idf doesn't have all the required submodules init'ed and updated. Please run `git submodule update --init --recursive` in both the `esp-idf` and `micropython`.

Or, to get exactly the minimum set of modules you need:
in micropython/ports/esp32: `make submodules`
in esp-idf:  `git submodule update --init components/json/cJSON components/esp32/lib components/esptool_py/esptool components/expat/expat components/lwip/lwip components/mbedtls/mbedtls components/micro-ecc/micro-ecc components/nghttp/nghttp2 components/nimble components/bt`

petrkr, 2020-01-17:
Hi jimmo,
I can confirm this way how to update `git submodule update --init --recursive` modules really works, at esp-idf it downloaded that `nimble` submodule and it compiles now.

I just wonder why it does not work when doing `git clone --recurse-submodules` and `git checkout v3.3.1 --recurse-submodules` and `git submodules update --recursive`

I have to learn about that submodules and how works these submodules in submodules.

closed 2020-01-17 by petrkr