micropython/micropython #5545: machine.RTC.init throws exception on Pyboard 1.0

peterhinch, 2020-01-16:
The code example in [the docs](http://docs.micropython.org/en/latest/library/machine.RTC.html) fails with a surprising traceback.
```python
rtc = machine.RTC()
rtc.init((2014, 5, 1, 4, 13, 0, 0, 0))
```
Outcome:
```
MicroPython v1.12-61-g0f16eeab2 on 2020-01-16; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> from machine import RTC
>>> rtc = RTC()
>>> rtc.init((2014, 5, 1, 4, 13, 0, 0, 0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 1 positional arguments but 2 were given
```

MrSurly, 2020-01-16:
That is rather surprising.  The code shows it only accepts `self`:
https://github.com/micropython/micropython/blob/master/ports/stm32/rtc.c#L465

peterhinch, 2020-01-17:
Is this a documentation issue? The undocumented **rtc.datetime** method appears able to set the RTC (tested on a Pyboard 1.x and ESP8266).

dpgeorge, 2020-01-20:
Note that, on the stm32 port, `rtc.init()` doesn't take any arguments beyond self, so if you pass one in then that's 2 arguments (self + argument) and so that's why the error message is as it is (related to the issue raised in #1893).

> Is this a documentation issue?

Yes it would appear so.

peterhinch, 2020-01-20:
I'll raise a PR against the doc.

peterhinch, 2020-01-21:
See https://github.com/micropython/micropython/pull/5553

kesterlester, 2021-02-06:
The example code here https://docs.micropython.org/en/latest/library/machine.RTC.html
still contains a suggestion to do:
```
rtc = machine.RTC()
rtc.init((2014, 5, 1, 4, 13, 0, 0, 0))
print(rtc.now())
```
which still fail (on my pyboard D-series) with the
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 1 positional arguments but 2 were given
```
message reported a year ago earlier in this thread.  Does that mean that no changes were made to the docs?

robert-hh, 2021-02-07:
It looks like the documentation is indeed wrong and the implementation is different to the other ports. on STM32
- rtc.init() takes no arguments
- rtc.now() does not exists and
- rtc.datetime() is provided to getting and setting date and time.