micropython/micropython #5547: py/compile: compile built-in with 'single' option is not CPython compatible

tve, 2020-01-17:
In CPython:
```
Python 3.8.0 (default, Oct 23 2019, 18:51:26)
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> eval(compile("4+5", "<string>", "single"))
9
>>>
```
In MicroPython (1.12):
```
MicroPython v1.12-68-g3032ae115-dirty on 2020-01-15; ESP32 module with ESP32
Type "help()" for more information.
>>> 4+5
9
>>> eval(compile("4+5", "<string>", "single"))
>>>
```

dpgeorge, 2020-01-20:
Thanks for the report, I can confirm the incompatibility.

It seems that the "single" option to compile in CPython is acting just like it would if evaluated at the REPL and printing the result of the expression that it executes/evaluates.  It doesn't actually return anything (just `None`).  Eg:
```python
x = eval(compile('if 1:\n 2\n 3', 'file', 'single'))
print(x)
```
This will print 2 then 3 during execution, but `x`, the result of the evaluation, is `None`.

To make MicroPython compatible it would need to print the results of expressions evaluated in a "single".

How did you come across this difference?  It doesn't seem like a common thing to do.

tve, 2020-01-20:
I'm trying to implement a repl-like facility using mqtt messages and I thought that `compile(.., 'single')` would be the most appropriate. Right now I'm doing:
```
        did_exec = False
        try:
            op = compile(cmd, "<mqtt>", "eval")
        except SyntaxError:
            did_exec = True
            op = compile(cmd, "<mqtt>", "exec")
        result = eval(op, globals(), None)
```
Coupled with a dupterm lets me collect the stdout. To be honest, I'm not sure what I really want in the end, the fact that expressions don't produce a result with `single` is not great. A quick fix would be to add a note to the docs.