micropython/micropython #5549: Pyboard D RTC calibration does not work

peterhinch, 2020-01-18:
On a Pyboard 1.1 the following works:
```python
>>> from pyb import RTC
>>> rtc=RTC()
>>> rtc.calibration()
0
>>> rtc.calibration(1)
>>> rtc.calibration()
1
>>> rtc.calibration(-1)
>>> rtc.calibration()
-1
```
On a Pyboard D SF2W and on an SF6W I get
```python
>>> from pyb import RTC
>>> rtc=RTC()
>>> rtc.calibration()
0
>>> rtc.calibration(1)
>>> rtc.calibration()  # +ve values seem to be negative sign extended
-511
>>> rtc.calibration(-1)  # It likes -ve values
>>> rtc.calibration()
-1
>>> 
```
Actual RTC behaviour matches the reported values - i.e. passing 1 yields a frequency which is substantially in error. This behaviour is odd as the definition of the RTC_CALR register appears identical for all three chips.

While the examples I have show good RTC performance the data for the SiT1533 oscillator indicates that calibration may sometimes be required (up to 20ppm error).

dpgeorge, 2020-01-20:
This was very recently fixed in e3187b052f14872fdb5e2d2338d359013a544fae (at least that was the intention of that commit).

It'd be good to know if the latest version works for you.

peterhinch, 2020-01-20:
This works on SF2W and SF6W: the problem reported above is fixed, and calibration has the desired effect.

closed 2020-01-20 by peterhinch