micropython/micropython #5550: can't build micropython: py/objmodule.c on moduledefs.h

sinc379, 2020-01-18:
I rearranged the locations to build to accomodate a port for micropython.
However, when I build, the generated (by makemoduledefs.py) moduledefs.h on build/genhdr/moduledefs.h generates this error:
build/genhdr/moduledefs.h:5:39: error: expected ‘}’ before ‘{’ token
#define MODULE_DEF_MP_QSTR_UARRAY { MP_ROM_QSTR(MP_QSTR_uarray), MP_ROM_PTR(&mp_module_uarray) },
build/genhdr/moduledefs.h:11:36: note: in expansion of macro ‘MODULE_DEF_MP_QSTR_UARRAY’
 #define MICROPY_REGISTERED_MODULES MODULE_DEF_MP_QSTR_UARRAY
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~
../py/objmodule.c:236:5: note: in expansion of macro ‘MICROPY_REGISTERED_MODULES’
     MICROPY_REGISTERED_MODULES
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
../py/objmodule.c:138:60: note: to match this ‘{’
 STATIC const mp_rom_map_elem_t mp_builtin_module_table[] = {
                                                            ^
make: *** [../py/mkrules.mk:47: build/py/objmodule.o] Error 1
What am I doing wrong here?
Thanks

dpgeorge, 2020-01-20:
> I rearranged the locations to build to accomodate a port for micropython.

You'll need to provide details of exactly what you did.  If you build MicroPython in the standard way then it should work.

sinc379, 2020-01-20:
I moved the folders to a directory upper, so everthing matches my custom Makefile. However, for some reason, it fails to compile with gcc like this.

stinos, 2020-01-21:
That's still not enough information I'm afraid, it won't be possible to provide you any help without knowing exactly what your build is doing. What directories were moved where, can you show the directory layout? What does your custom Makefile do? Does everything still build using the original Makefile?