micropython/micropython #5552: micropython-lib version included in official downloads is unclear

jaseg, 2020-01-20:
The [official downloads page](http://micropython.org/download) lists pre-built images by micropython version, but does not give any information on which micropython-lib version is used. I've had a few issues where I got stack traces into micropython-lib but was unable to find the matching source version.

I think either the micropython-lib repo should contain tags pointing to the corresponding micropython versions, or the micropython-lib version/commit/tag should be linked on the downloads page and somewhere in the micropython repo.

Thanks!

peterhinch, 2020-01-20:
The official library is [here](https://github.com/micropython/micropython-lib) and pre-built images should include code from there. Can you identify a specific instance where this is not the case?

jaseg, 2020-01-20:
Thank you for your response. I found that repo, but my problem is that there is no link between a certain micropython version and a particular commit in that repo.

Say I download micropython v1.9.3 from November 2017. If I go to the [lib repo](https://github.com/micropython/micropython-lib) there, the latest commit on the master branch is from February 2019, newer than my micropython version. What I'd like to have is something (e.g. the downloads page) telling me which lib version my micropython version is using.

Specifically, I got an error in urequests on micropython v1.12, and the line numbers did not match up with the micropython-lib repo's master branch.