micropython/micropython #5554: ESP32: upip, wifi: bcn_timout,ap_probe_send_start

the-stanely, 2020-01-21:
Using Heltec ESP32 WIFI Kit.  Can't run upip for any installs because WiFI disconnects shortly after package install started.  Googling for the issue, I found a post by ESP_Sprite that seems a plausible explanation...

>>> The issue is that most of the ESP32 is 'locked up' while doing flash writes; the architecture is somewhat dependent on being able to read code from the flash chip on the fly, and while doing a write it cannot do this, so it will suspend most tasks until the write is over. If you do a lot of large writes back-to-back, this means a bunch of tasks won't run for a long time, and the task watchdog kicks in. Suggest doing a vTaskDelay(10) or something between flash writes, to give other tasks the chance to run.

I don't know if this is a Heltec, ESP32, or upip problem.  How would I go about trying to add the delay to upip?

dpgeorge, 2020-01-22:
Thanks for the report, I can reproduce the problem using `upip.install('logging')` after connecting to the wifi.

The problem doesn't seem to be present with IDF 3.3.  The problem looks very similar to this ESP IDF issue: https://github.com/espressif/esp-idf/issues/4479

the-stanely, 2020-01-22:
I looked at IDF 3.3 on Espressif's site and don't understand how to incorporate it into my Mu & MicroPython development setup.

the-stanely, 2020-01-22:
It seems like ESP IDF is more like the Arduino-IDE approach to development.  Any upip is done in the local host environment, and the entire project is flashed to the ESP32, so you never do a REPL upip on the target.  The way MicroPython with Mu works is that the development environment is on the target, so interpreter commands need to work on the target.  This is where the problem is. 

The only way that espressif/esp-idf#4479 might be similar is if they're doing flash writes which causes the wifi beacon to time out.  I think that's the crux of this problem.

dpgeorge, 2020-01-22:
> I looked at IDF 3.3 on Espressif's site and don't understand how to incorporate it into my Mu & MicroPython development setup.

If you revert to commit c14ff6194c224190c7e6787a2bffbdbb495be4b4 then it uses IDF v3.3.

the-stanely, 2020-01-22:
I'm using this binary, esp32-idf3-20200122-v1.12-68-g3032ae115.bin, and I think it says idf3.  I haven't tried the idf4 but will right away.  I don't have esp idf installed and wouldn't be able to build a binary.  I din't see any binaries in that commit.

the-stanely, 2020-01-22:
Thank you!  I loaded this binary: esp32-idf4-20200122-v1.12-68-g3032ae115.bin.  On boot it reports:  esp32-idf4-20200122-v1.12-68-g3032ae115.bin.  upip is working with this firmware.  upip did not work with: esp32-idf3-20200122-v1.12-68-g3032ae115.bin

the-stanely, 2020-01-23:
I just realized that the latest binary looks like a daily build.  So the one that was not working was an earlier version.  I still have it on my disk.  It was: esp32-idf3-20200117-v1.12-68-g3032ae115.bin

I'm sorry for my confusion.

petrkr, 2020-04-09:
On latest build 2020 04 09 still same issue
`bcn_timout,ap_probe_send_start`

latest build from december (with SDK 3.3) can not reproduce.. with SDK 3.3.1 its like 50/50...

also there are more AP which causes this and for example AP from UPC ISP does not have problem even with 3.3.1 at all

petrkr, 2020-04-09:
Consider try 3.3.2 ?

I tried compile it but they changed lot of things including BLE (which maybe could be good reason switch to it ASAP to be upstream)

```
LINK build-GENERIC/application.elf
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_gatts.o):(.literal.ble_gatts_start+0x10): undefined reference to `nimble_platform_mem_malloc'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_gatts.o): In function `ble_gatts_free_mem':
ble_gatts.c:(.text.ble_gatts_free_mem+0xa): undefined reference to `nimble_platform_mem_free'
ble_gatts.c:(.text.ble_gatts_free_mem+0x17): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_gatts.o): In function `ble_gatts_free_svc_defs':
ble_gatts.c:(.text.ble_gatts_free_svc_defs+0x8): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_gatts.o): In function `ble_gatts_start':
ble_gatts.c:(.text.ble_gatts_start+0x3f): undefined reference to `nimble_platform_mem_malloc'
ble_gatts.c:(.text.ble_gatts_start+0x67): undefined reference to `nimble_platform_mem_malloc'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(esp_nimble_hci.o):(.literal.ble_buf_free+0x10): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(esp_nimble_hci.o):(.literal.esp_nimble_hci_init+0x24): undefined reference to `nimble_platform_mem_calloc'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(esp_nimble_hci.o): In function `ble_buf_free':
esp_nimble_hci.c:(.text.ble_buf_free+0xd): undefined reference to `nimble_platform_mem_free'
esp_nimble_hci.c:(.text.ble_buf_free+0x1a): undefined reference to `nimble_platform_mem_free'
esp_nimble_hci.c:(.text.ble_buf_free+0x27): undefined reference to `nimble_platform_mem_free'
esp_nimble_hci.c:(.text.ble_buf_free+0x34): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(esp_nimble_hci.o): In function `esp_nimble_hci_init':
esp_nimble_hci.c:(.text.esp_nimble_hci_init+0x10): undefined reference to `nimble_platform_mem_calloc'
esp_nimble_hci.c:(.text.esp_nimble_hci_init+0x20): undefined reference to `nimble_platform_mem_calloc'
esp_nimble_hci.c:(.text.esp_nimble_hci_init+0x30): undefined reference to `nimble_platform_mem_calloc'
esp_nimble_hci.c:(.text.esp_nimble_hci_init+0x40): undefined reference to `nimble_platform_mem_calloc'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(os_msys_init.o): In function `os_msys_buf_alloc':
os_msys_init.c:(.text.os_msys_buf_alloc+0x8): undefined reference to `nimble_platform_mem_calloc'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(os_msys_init.o): In function `os_msys_buf_free':
os_msys_init.c:(.text.os_msys_buf_free+0x8): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_att_svr.o): In function `ble_att_svr_free_start_mem':
ble_att_svr.c:(.text.ble_att_svr_free_start_mem+0x8): undefined reference to `nimble_platform_mem_free'
build-GENERIC/esp-idf/bt_nimble/libbt_nimble.a(ble_att_svr.o): In function `ble_att_svr_start':
ble_att_svr.c:(.text.ble_att_svr_start+0x1b): undefined reference to `nimble_platform_mem_malloc'
make: *** [Makefile:777: build-GENERIC/application.elf] Error 1
```

tve, 2020-04-09:
I have seen these symptoms using esp-idf v4 when it runs out of memory during upip. Upip ends up using SSL, which is pretty memory intensive and esp-idf runs out of heap. Make sure you start from a ctrl-d then connect to wifi, then upip and nothing else. If that still doesn't work you need to copy files manually, you can run upip on the host. I'm working on some fixes for this, but it's not going to be instantaneous.

petrkr, 2020-04-09:
it's clean install without anything run yet.
tested case like

1) erase flash
2) flash
3)
import network
n=network.WLAN()
n.active(1)
n.connect(ssid,psk)

import upip
upip.install()

sometimes it goes sometimes not on 3.3.1 SDK, but it works always on 3.3

Also it's documented issue on 3.3.1 SDK. Maybe consider update to 3.3.2 ASAP before implementation of BLE will be too much different so it would not be easily possible

the-stanely, 2020-04-09:
Are you sure it's running out of heap, and not wifi beacon tiemout bc of extended flash writes?

tve, 2020-04-09:
> Are you sure it's running out of heap, and not wifi beacon tiemout bc of extended flash writes?

That might be possible too. I did trace the (very repeatable) failure I had down into ESP-IDF's threads and I'm pretty sure I saw a heap allocation failure, but now I'm not 100% sure. It did disconncet wifi and reconnect as a result.

If you have a module with PSRAM you could try using that and see a comparison, or if you build from scratch you could divide the MP heap size in half here https://github.com/micropython/micropython/blob/master/ports/esp32/main.c#L103 thereby leaving more for esp-idf. If those things don't fix the issue then it's not out of memory, if they do fix the issue then memory may play a role. In either case, please report back if you run the experiment!

petrkr, 2020-04-10:
Old works everytime
```
I (476) cpu_start: ESP-IDF:          v3.3
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
114256
>>> gc.collect()
>>> gc.mem_free()
114320
```

Latest downloadable, works at first, when CTRL+D starts to do bcm timeout, even hard reset (button) does not help.. Sometimes will help power off (by usb cable) and on again

```
I (551) cpu_start: ESP-IDF:          v3.3.1
MicroPython v1.12-356-gd6f80963d on 2020-04-09; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
109888
>>> gc.collect()
>>> gc.mem_free()
110144

I (98131) modsocket: Initializing
Installing shutil
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing micropython-shutil 0.0.3 from https://micropython.org/pi/shutil/shutil-0.0.3.tar.gz
I (106871) wifi: bcn_timout,ap_probe_send_start
I (109381) wifi: ap_probe_send over, resett wifi status to disassoc
I (109381) wifi: state: run -> init (c800)
I (109381) wifi: pm stop, total sleep time: 11014517 us / 22533291 us

>>> gc.mem_free()
100592
```

If I will do on latest GIT 
 size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
replaced by
 size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)/2;

then it works everytime, not mather how many times CTRL+D, or reset button will do

```
I (554) cpu_start: ESP-IDF:          v3.3.1
MicroPython v1.12-359-g61bf24b76-dirty-build-octopusLAB on 2020-04-10; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> import gc
>>> gc.mem_free()
53600
```

Based on this it would be connected with that heap space, but strange is December version has more free RAM than latest for python and it works... It could be connected also by change to another SSL library

petrkr, 2020-04-10:
Seems if I will do
```
diff --git a/ports/esp32/main.c b/ports/esp32/main.c
index 81e4a6434..f08577244 100644
--- a/ports/esp32/main.c
+++ b/ports/esp32/main.c
@@ -100,7 +100,7 @@ void mp_task(void *pvParameter) {
     }
     #else
     // Allocate the uPy heap using malloc and get the largest available region
-    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
+    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)-2048;
     void *mp_task_heap = malloc(mp_task_heap_size);
     #endif
 ```
then it works.

if I will do less like `-1024` or even `-1536` then still not enough, but `-2048` seems to be stable.. Atlast at mine setup

tve, 2020-04-10:
Thanks for testing!!! I know of a way to free ~1400 bytes of RAM when using SSL. That may not be quite enough, will have to think a bit. Maybe we just have to bite the bullet and hard-code some subtraction from the max heap size like you did.

> strange is December version has more free RAM than latest for python and it works

I'm not sure how to read your statement, but ESP-IDF is running out of the "C" heap, which contains all of memory. The python heap is separate, it's what you reduced in your tests, and it's not running out by a long shot. I hope this clarifies...

petrkr, 2020-04-10:
you can try `http://iot.petrkr.net/fw/micropython-740946736-octopus-908d33160-20200410-test.bin`
it's latest master with our octopus initial setup for our platform.. I just using this tree (check at https://github.com/petrkr/micropython/commits/octopus)

petrkr, 2020-04-14:
you can try 3.3.2 SDK Build `http://iot.petrkr.net/fw/micropython-octopus-20200414-3.3.2-test.bin`

https://github.com/petrkr/micropython/commits/octopus-3.3.2

dpgeorge, 2020-04-14:
The IDF has been updated to v3.3.2, which may help to prevent this issue.  Although, if it's related to RTOS memory fragmentation, then on IDF v4 the issue could still be there, and it may come back again in newer IDF versions.

petrkr, 2020-04-19:
Now for me this issue appear again when I turn on BLE together with WiFi.

I can confirm with 3.3.1 it do aswell. So 3.3.2 makes it much better in WiFi-only setup.

Also I can confirm, if I will use my "minus memory trick" which I used in SDK 3.3.1. Then it is much better. But here I have to remove 8kB.. If I will do just 4kB then it appear again within like 10 minutes.

Also there is some problem when I will turn on WiFi with BLE it seems to mess up BLE because it scan much less devices during gap_scan()

I solved that by adding these lines to sdk config
```
# Coexistent BLE + WiFI
BTDM_COEX_BT_OPTIONS=y
BTDM_COEX_BLE_ADV_HIGH_PRIORITY=y
```

memory 'hack'
```
-    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT);
+    size_t mp_task_heap_size = heap_caps_get_largest_free_block(MALLOC_CAP_8BIT)-4096*2;
```

tom-van, 2021-05-03:
I got the same problem with esp32s2 port. C heap is exhausted by tusb_task and WiFi disconnects during WebREPL connect.
To confirm this is really malloc() failure I added ESP_LOGE() to esp-idf/components/esp_wifi/esp32s2/esp_adapter.c
See the log:

`ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x11e4
load:0x4004c000,len:0x8f0
load:0x40050000,len:0x293c
entry 0x4004c1b0
I (428) wifi:wifi driver task: 3fff8268, prio:23, stack:6656, core=0
I (428) wifi:wifi firmware version: 0aebefd
I (428) wifi:wifi certification version: v7.0
I (428) wifi:config NVS flash: enabled
I (428) wifi:config nano formating: disabled
I (438) wifi:Init data frame dynamic rx buffer num: 32
I (438) wifi:Init management frame dynamic rx buffer num: 32
I (448) wifi:Init management short buffer num: 32
I (448) wifi:Init dynamic tx buffer num: 32
I (448) wifi:Init static rx buffer size: 1600
I (458) wifi:Init static rx buffer num: 10
I (458) wifi:Init dynamic rx buffer num: 32
I (598) wifi:mode : sta (xxx)
I (598) wifi:enable tsf
connecting to network...
I (1218) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2538) wifi:state: init -> auth (b0)
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2548) esp_adapter: wifi_malloc: no memory
I (2558) wifi:state: auth -> assoc (0)
I (2568) wifi:state: assoc -> run (10)
I (2578) wifi:connected with xxxx, aid = 3, channel 6, BW20, bssid = xxxx
I (2578) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (2578) wifi:pm start, type: 1
 
W (2608) wifi:<ba-add>idx:0 (ifx:0, xxx), tid:6, ssn:1, winSize:64
I (2628) wifi:AP's beacon interval = 102400 us, DTIM period = 2
network config: ('192.168.xx', '255.255.255.0', '192.168.xx', '192.168.xx')
WebREPL daemon started on ws://192.168.xx:8266
Started webrepl in normal mode
MicroPython v1.15-58-g9bb7e7057-dirty on 2021-05-03; ESP32S2 module with ESP32S2
Type "help()" for more information.
>>> W (45248) wifi:<ba-add>idx:1 (ifx:0, 64:66:b3:1d:a1:12), tid:0, ssn:1, winSize:64
E (45258) esp_adapter: wifi_malloc: no memory
E (45258) esp_adapter: wifi_malloc: no memory
`
  ... about 50 failed malloc() ...

`E (51788) esp_adapter: wifi_malloc: no memory
I (51848) wifi:bcn_timout,ap_probe_send_start
W (51848) wifi:m f probe req l=0`

Now it looks clear!
WTF why ESP-IDF WiFi is silent when runs out of heap???!!!!

robert-hh, 2021-05-03:
Thank you for the note. I had big problems with WiFi and the esp32-s2. Reducing the amount of memory allocated for the python heap as mentioned above solved that.

jonnor, 2024-10-01:
upip has been replaced by mip in MicroPython 1.20, https://github.com/micropython/micropython/releases/tag/v1.20.0

Furthermore there has been many improvements over the last years in ESP32 WiFi stability, as well as other fixes from esp-idf. Please test again on a recent MicroPython, and re-open issue with more details.

closed 2024-10-01 by jonnor