micropython/micropython #5556: NUCLEO_H743ZI fails Error 1

K2AAE, 2020-01-21:
After two hours of screwing around I remember why I hate Linux: 

Using Windows 10 Linux for Windows I installed Ubuntu 18.04, updates, upgraded, installed build_essentials and arm-linux-gnueabi-gcc (ubuntu default) did make submodules then make and after I figured out where to change the compiler prefix I got:

CC ../../lib/embed/abort_.c
CC ../../lib/utils/printf.c
MPY dht.py
MPY lcd160cr.py
MPY lcd160cr_test.py
MPY onewire.py
GEN build-NUCLEO_H743ZI/frozen_content.c
CC build-NUCLEO_H743ZI/frozen_content.c
CC ../../lib/libc/string0.c
CC ../../lib/mp-readline/readline.c
CC ../../lib/netutils/netutils.c
CC ../../lib/netutils/trace.c
CC ../../lib/netutils/dhcpserver.c
CC ../../lib/timeutils/timeutils.c
CC ../../lib/utils/pyexec.c
CC ../../lib/utils/interrupt_char.c
CC ../../lib/utils/sys_stdio_mphal.c
CC ../../lib/utils/mpirq.c
CC ../../lib/libm_dbl/__cos.c
In file included from ../../lib/libm_dbl/__cos.c:51:0:
../../lib/libm_dbl/libm.h:93:8: error: conflicting types for ‘__sin’
 double __sin(double, double, int);
        ^~~~~
In file included from /usr/arm-linux-gnueabi/include/features.h:424:0,
                 from /usr/arm-linux-gnueabi/include/bits/libc-header-start.h:33,
                 from /usr/arm-linux-gnueabi/include/stdint.h:26,
                 from /usr/lib/gcc-cross/arm-linux-gnueabi/7/include/stdint.h:9,
                 from ../../lib/libm_dbl/libm.h:15,
                 from ../../lib/libm_dbl/__cos.c:51:
/usr/arm-linux-gnueabi/include/bits/mathcalls.h:64:1: note: previous declaration of ‘__sin’ was here
 __MATHCALL_VEC (sin,, (_Mdouble_ __x));
 ^
In file included from ../../lib/libm_dbl/__cos.c:51:0:
../../lib/libm_dbl/libm.h:94:8: error: conflicting types for ‘__cos’
 double __cos(double, double);
        ^~~~~
In file included from /usr/arm-linux-gnueabi/include/features.h:424:0,
                 from /usr/arm-linux-gnueabi/include/bits/libc-header-start.h:33,
                 from /usr/arm-linux-gnueabi/include/stdint.h:26,
                 from /usr/lib/gcc-cross/arm-linux-gnueabi/7/include/stdint.h:9,
                 from ../../lib/libm_dbl/libm.h:15,
                 from ../../lib/libm_dbl/__cos.c:51:
/usr/arm-linux-gnueabi/include/bits/mathcalls.h:62:1: note: previous declaration of ‘__cos’ was here
 __MATHCALL_VEC (cos,, (_Mdouble_ __x));
 ^
In file included from ../../lib/libm_dbl/__cos.c:51:0:
../../lib/libm_dbl/libm.h:95:8: error: conflicting types for ‘__tan’
 double __tan(double, double, int);
        ^~~~~
In file included from /usr/arm-linux-gnueabi/include/features.h:424:0,
                 from /usr/arm-linux-gnueabi/include/bits/libc-header-start.h:33,
                 from /usr/arm-linux-gnueabi/include/stdint.h:26,
                 from /usr/lib/gcc-cross/arm-linux-gnueabi/7/include/stdint.h:9,
                 from ../../lib/libm_dbl/libm.h:15,
                 from ../../lib/libm_dbl/__cos.c:51:
/usr/arm-linux-gnueabi/include/bits/mathcalls.h:66:1: note: previous declaration of ‘__tan’ was here
 __MATHCALL (tan,, (_Mdouble_ __x));
 ^
../../lib/libm_dbl/__cos.c:61:8: error: conflicting types for ‘__cos’
 double __cos(double x, double y)
        ^~~~~
In file included from /usr/arm-linux-gnueabi/include/features.h:424:0,
                 from /usr/arm-linux-gnueabi/include/bits/libc-header-start.h:33,
                 from /usr/arm-linux-gnueabi/include/stdint.h:26,
                 from /usr/lib/gcc-cross/arm-linux-gnueabi/7/include/stdint.h:9,
                 from ../../lib/libm_dbl/libm.h:15,
                 from ../../lib/libm_dbl/__cos.c:51:
/usr/arm-linux-gnueabi/include/bits/mathcalls.h:62:1: note: previous declaration of ‘__cos’ was here
 __MATHCALL_VEC (cos,, (_Mdouble_ __x));
 ^
../../py/mkrules.mk:47: recipe for target 'build-NUCLEO_H743ZI/lib/libm_dbl/__cos.o' failed
make: *** [build-NUCLEO_H743ZI/lib/libm_dbl/__cos.o] Error 1
kevin@Aorus:/mnt/d/git2/ports/stm32$

rolandvs, 2020-01-21:
I have tested it with the current repo and it compiles fine.
See last lines... However some of the embedded linux platforms use `linaro gcc arm` tools instead of ubuntu's distribution to cross compile the Yocto/Bootlin stuff... Maybe that helps in your config.

```
make BOARD=NUCLEO_F743ZI all
....
LINK build-NUCLEO_H743ZI/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 460308	    184	  54104	 514596	  7da24	build-NUCLEO_H743ZI/firmware.elf
GEN build-NUCLEO_H743ZI/firmware.dfu
GEN build-NUCLEO_H743ZI/firmware.hex
develop@ubuntu:~/workspace/micropython/ports/stm32$ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
```

K2AAE, 2020-01-21:
This is what I Google Fu'ed 
sudo apt install gcc-arm-linux-gnueabi g++-arm-linux-gnueabi  
Is there another gnu for ARM? 
Thanks 
Kevin A

rolandvs, 2020-01-21:
https://www.linaro.org/downloads/

jimmo, 2020-01-22:
gcc-arm-linux-gnueabi is what you'd use for compiling programs to run on an ARM host running linux.

You need the baremetal compiler -- gcc-arm-none-eabi

K2AAE, 2020-01-22:
The sound you just heard was my head exploding... I guess gcc-arm-linux-gnueabihf is the tools I need for STM32MPU1.. That also explains why I had to change the compiler prefix
Thank you all

K2AAE, 2020-01-22:
After a dinner break:
arm-none-eabi-size build-NUCLEO_H743ZI/firmware.elf
   text    data     bss     dec     hex filename
 458840     184   54104  513128   7d468 build-NUCLEO_H743ZI/firmware.elf
GEN build-NUCLEO_H743ZI/firmware.dfu
arm-none-eabi-objcopy -O binary -j .isr_vector build-NUCLEO_H743ZI/firmware.elf build-NUCLEO_H743ZI/firmware0.bin
arm-none-eabi-objcopy -O binary -j .text -j .data build-NUCLEO_H743ZI/firmware.elf build-NUCLEO_H743ZI/firmware1.bin
python3 ../../tools/dfu.py -b 0x08000000:build-NUCLEO_H743ZI/firmware0.bin -b 0x08040000:build-NUCLEO_H743ZI/firmware1.bin build-NUCLEO_H743ZI/firmware.dfu
GEN build-NUCLEO_H743ZI/firmware.hex
arm-none-eabi-objcopy -O ihex build-NUCLEO_H743ZI/firmware.elf build-NUCLEO_H743ZI/firmware.hex
kevin@Aorus:/mnt/d/git2/ports/stm32$
Had to remember make clean.....

closed 2020-01-22 by K2AAE