micropython/micropython #5561: NUCLEO-H743ZI DFU mode failure

K2AAE, 2020-01-22:
I can not get the H743ZI into DFU mode nor can I find anything about how to put the device into DFU mode other then ST-LINK V3 can do it if you have ST-LINK V3... I've read the information about how to generate other firmware BUT last night Microsoft upgraded my system with WSL Version 2, this morning I tried to get WSL Version 2 to work and not only failed but killed version 1, no Linux, nothting but the files I generated last night..

rolandvs, 2020-01-22:
I assume that you use the second USB connection for this purpose?

rolandvs, 2020-01-22:
Just to inform you beforehand, (wired) Ethernet is not working (yet).

K2AAE, 2020-01-22:
On the bottom of the board they labeled the connector BT0 da.... right above it is VDD - I did find Micropython in MBED but: Error: /src/micropython/modmachine.c:38:10: fatal error: 'cmsis.h' file not found
#include "cmsis.h"
         ^~~~~~~~~
1 error generated.

My luck, have to go find the animals then I'll try the DFU again, their CUBE programmer seems nice but then the ST-LINK utility is nice.. Thank you for the answer !

rolandvs, 2020-01-22:
Another time saver:

Are you sure you have a NUCLEO-H743ZI or do you have a NUCLEO-H743ZI2? The mpy port is for the *ZI and the *ZI2 needs some minor changes (like the three LEDs).

The *ZI2 does not work using DFU as I just tested and STLINK is not working also as there is an ST-LINK V3 on the *ZI2 board.

So the only way (now) is to use an external tool like STM32CubeProgrammer or STM32Programmer_CLI to program the hex/bin file directly.

Watch this space for an update.

rolandvs, 2020-01-22:
`NUCLEO_H743ZI2`

NOTE: MPY does not work, probably clock related and some GPIO changes compared to the *ZI board. USB storage and CDC are not working and the STLINK Virtual COM port is also not working.

For what its worth: In order to program the `NUCLEO_H743ZI2` with the `ST-LINK3` it is the quickest way to install the `STM32Cube_Programmer`. However on OSX it seems impossible to get the thing running due to some obscure JAVA related error. The GUI version will not work, but the CLI version will.

So next thing is to make some quick adjustments to the Makefile:

```
STFLASH3 ?= /Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI
```
and
```
deploy-stlink3: $(BUILD)/firmware.dfu
	$(ECHO) "Writing $(BUILD)/firmware0.bin and firmware1.bin to the board via ST-LINK v3"
	$(Q)$(STFLASH3) -c port=SWD -d $(BUILD)/firmware0.bin $(TEXT0_ADDR) -d $(BUILD)/firmware1.bin $(TEXT1_ADDR)
```

I tried to isolate the CLI related stuff from the rest and move it to `/usr/local/bin`, but somehow it will not work (yet).

Next thing is to verify if the image is loaded (using SEGGER JLINK :-) and start fixing...

K2AAE, 2020-01-22:
I've used DFU  loading before but the NUCLEO boards are all not doing DFU: G031K8, L031K6, G431KB STLINK-V3,G431RB STLINK-V3, L452RE, L476RG, F401RE and of course the H743ZI the first version with STLINK2 
None of those boards are seen by STMCUBE Programmer so there is something going on... too bad I killed WSL this morning I could try the python loader. The latest NUCLEO-G431RB board actually has a jumper for BOOT0! 
On the Link 3 stuff, I think I read that link 3 supports drag and drop?  What would happen if I used https://www.st.com/en/development-tools/stsw-link004.html to write the firmware.dfu file?

rolandvs, 2020-01-22:
You can drop a file on it. However the firmware0.bin and firmware1.bin need to be packed into one blob and be padded to fill the gap between them. firmware0.bin gets programmed as it starts at 0x80000000...

K2AAE, 2020-01-23:
After wearing the letters off my keyboard it seems STM32CubeProg has a readme about DFU drivers, after you install the program and IT installs USB drivers you have to run a "batch" file but not until you removed something, that something is something they don't really talk about.... The batch file fails without errors but it does not install any INF files. I have to go visit the system error logs to see if I can trace down what is not loading or error. It would be a good thing with MicroPython could be drag and dropped since that seems to be where STM is going.

K2AAE, 2020-01-23:
I finally got WSL to work, seems they changed the install and the Google fu sites I found didn't have the fact you have to turn on Virtual Machine now and that means it's useless. I spent an hour waiting on the 4 minute compile, when it finished I used STM32CubeProg to write the HEX file to the NUCLEO-H743ZI after I used putty to connect. I'm suppose to be able to convert WSL back to version 1, if so I'll keep using it for these projects, it was fast, otherwise its buying a new linux box.

K2AAE, 2020-01-23:
I switched back to WSL 1, it took 3 minutes to build, WSL 2 is dead.

K2AAE, 2020-01-25:
ST replied,  my issue with DFU has been thinking the USB ST-Link connector is what you use to DFU. Nope, CN3 on the STM32H743ZI is plugged into the computer, BOOT0 to +3.3 then plug the connector into the ST-Link MicroUSB  for power, the DFU USB device shows in windows device manager. 
For loading Micropython I'll just use the STM32CubeProg via the ST-Link connector and download the HEX file, if the NUCLEO board doesn't have an USB connector for the device why break out a USB to UART... Somewhere in the many  application notes I've downloaded I saw a list of all the devices and what port they use for BOOT0, for some reason I thought that was for boards without built-in programmers...