micropython/micropython #5565: Run `.mpy` files from cpython?

traverseda, 2020-01-23:
It would be great if there was one unified way to write faster python extensions. For various reasons we can't really port cython to micropython, but what aboutproviding some way for cpython to run mpy extensions? #5083 seems very powerful.

Is there any way to do that? Perhaps is would be easier to provide some sort of bridge between micropython and cpython?

SaxonRah, 2020-07-20:
Hmm, not sure why you'd even consider porting Cython to MicroPython. To me that would imply that some part of MicroPython would be relying on what Cython is doing and therefore MicroPython would be come dependent on Cython. I don't think anyone wants that.

However, It does seem much more logical to get Cython to support MicroPython, making Cython have the MicroPython dependency. Which in turn could generate .mpy files or even elf files as you linked in #5083 In-fact I don't see why Cython wouldn't support MicroPython syntax already. Cython's scope seems like compiling MicroPython down would in fact be something they would want to support, as it aligns with much of what they are doing already. Although, this should probably be brought up to Cython devs not MicroPython devs anyways as they would be ultimately be the ones having to maintain it. If Cython supported MicroPython, this would mean you could write in C/C++/Cython/Python/MicroPython and have it compile down to a binary for embedded devices. This would be extremely useful and game changing.

I think if we wanted zero Cython dependency in both directions. The first step is to write a trans-piling layer. Where we can trans-compile (translate) Python to MicroPython and MicroPython back into Python. After that, essentially we have our own Cython-like  Python to MicroPython ecosystem for microcontrollers. From that point it'd be much more obtainable that Cython could statically optimize MicroPython.

kdschlosser, 2020-08-13:
This is a fantastic idea. 

Even tho the extension modules created using Cython are large you do get an enormous speed boost in the running of the code located in the extension module. On a desktop PC the speed increase is somewhere between 50 and 200 times faster compared to running the same code as "Pure Python". The truly neat thing you can do with Cython is it will take native python code using the normal python syntax and it will convert and compile it into an extension module. A person is able to get that speed boost without having to know a lick of C/C++.

If someone has not done so already I am going to see if I can possibly get the Cython developers attention with adding support for MicroPython. I have worked with them before with adding support for Stackless Python (a Python flavor), They added support for it. I think they might have interest in doing this if it doesn't require an enormity of work. 

I am sure that they will have questions that will need to be answered and I do not have enough MicroPython knowledge to be able to help with that respect. Where should I direct them to ask these questions?

kdschlosser, 2020-08-13:
@SaxonRah

I believe the OP is not referring to Cython but instead CPython. as I stated above being able to compile MicroPython C extensions using Cython would be a fantastic idea.

traverseda, 2020-08-13:
Basically I just think we need *some kind* of cross-platform compiled python support that works across multiple implementations, (cpython, pypy, micropython) and it seems like `.mpy` might be a good option for that? Alternatively it might be too closely tied to the micropython internals.

SaxonRah, 2020-08-13:
> @SaxonRah
> 
> I believe the OP is not referring to Cython but instead CPython. as I stated above being able to compile MicroPython C extensions using Cython would be a fantastic idea.
Yeah, but I really wanted to spark some conversation regarding his comment "For various reasons we can't really port cython to micropython" since there was not much about it any where else on the internet except a few posts on some forums, but nothing of great usefulness. It really is a good idea. I just wish i knew more about the internals of micropython to help. 

> Basically I just think we need _some kind_ of cross-platform compiled python support that works across multiple implementations, (cpython, pypy, micropython) and it seems like `.mpy` might be a good option for that? Alternatively it might be too closely tied to the micropython internals.
Maybe I'm missing something here, but wouldn't trans-piling be better? I ask this because python, micropython, pypy are all either subset languages of Cython or so similar. To add all said languages reside in the same language level (logic gates/machine code/assembly/C C++/Cython Python MicroPython/). So the language translation could be as simple as some regular expressions, or if needed fancier translation, use something like sly or ply for lexical analysis for simplicity. Also the cross-compiler 'mpy-cross' already has a device translation step that happens when a mpy is created. So wouldn't really all that be needed is to be able to trans-pile code from Cython / Python to MicroPython?

traverseda, 2020-08-13:
>I ask this because python, micropython, pypy are all either subset languages of Cython or so similar.

In the past you couldn't even run Cython modules under pypy. Last I checked Cython was very closely integrated with CPython "c api" which is itself *very* closely linked to the internals of the CPython implementation. This has gotten a lot better over time, you can read more about porting a Cython app to PyPy [here](https://cython.readthedocs.io/en/latest/src/userguide/pypy.html). Now PyPy is probably the most mature alternative python implementation, it's certainly one of the oldest that's still actively maintained.

In order to get Cython working for micropython in the same way it works for PyPy you'd need to implement a micropython equivalent to [cpyext](https://morepypy.blogspot.com/2018/09/inside-cpyext-why-emulating-cpython-c.html), which I think would be quite challenging. I'm not sure that it's doable inside the confines of a microcontroller, but I don't really know.

I think any effort to make Cython a truly multi-interpreter solution has to be pushed from inside Cython itself. Implementing Cython compatibility is *hard*. It might be easier for Cython to implement compatibility with other interpreters?

---

That's why I think approaching it from the micropython angle might be easiest. It probably has a much smaller "Python.h" that would need to be implemented by other interpreters, and in general a much smaller ABI. I'm not sure that works in practice though.

kdschlosser, 2020-08-14:
Cython has Stackless Python support built into it. While Stackless Python is extremely close to CPython there still needed to be changes made to Cython in order for extension modules compiled with Cython to run on Stackless Python.

Now if possible a module compiled using Cython to run on MicroPython would never be able to be run in CPython. this is because of the special hardware considerations in MicroPython. To run a module compiled with Cython for CPython would only be able to run on MicroPython if MicroPython has 100% of the CPython API in tact,. I do not believe that MicroPython has kept all of the original CPython API but I could be wrong with that assumption.

jonnor, 2024-07-27:
On the bindings level, the MicroPython implementation is completely different from CPython. So there would be considerable work to translate this in a generic manner.
If someone wanted to implement a CPython library that can load MicroPython modules, and provide interoperability that way, that would be great. But I do not think that this is in scope for the MicroPython core project. One strategy would be to embed a MicroPython interpreter, which MicroPython has good support for - see the "embed" port and examples in the repo.

In the meantime - a practical way to run MicroPython code on CPython is to use a subprocess. Either as a one-off process with stdin/stdout communication. Or as a long-lived service, with communication going over sockets et.c.

kdschlosser, 2024-07-27:
The C API for CPython and MicroPython are completely different. They may as well be on opposite ends of the universe from each other. It would be a fantastic amount of work that would be needed to make it possible. Not really worth the effort to do.

Now... Just to let you know... There has been written a little nugget of code that will read C header files and write the code needed to make that C code into an extension module for MicroPython. It makes it a lot easier when wanting to use a C library in MicroPython. As an example, C libraries like libJPEG can be used inside of MicroPython very easily thanks to that script...

KeithCu, 2025-02-23:
I think the translation to make a Cython C code work with MicroPython could be handled without changing Cython and with minimal changes to C and no changes to MicroPython. You just need a magic header file. Here's some sample code I coaxed out of Grok 3.

I'm going to keep working on it because I want to run Cython with MicroPython, but it would be great if people with actual knowledge were interested ;-) MicroPython + (compiled Cython) is a great possibility to bring Python to new places.

[micropython.txt](https://github.com/user-attachments/files/18928835/micropython.txt)

jonnor, 2025-02-23:
@KeithCu this looks like the start of a CPython C API compatibility layer on top of MicroPython C API? Just trying to understand the overall approach. How would one use this with Cython?

I could be interested in testing and giving feedback on a Cython for MicroPython, as it would be relevant for data science / numeric workflows.

sosi-deadeye, 2025-02-23:
What if `CPython` runs a `Micropython` VM and the Micropython-VM loads the `mpy`-files? This should be easier to implement, instead of creating something new.

KeithCu, 2025-02-23:
@jonnor What someone could do is is write a Cython application as per usual, and then replace the Python header in the outputted C files with this header, so it uses the MicroPython APIs. What do you think?

I created a repo:
https://github.com/KeithCu/Cython2Upy

kdschlosser, 2025-02-23:
What sparks my interest the most about using Cython and MicroPython is not having the ability to write Cython code and have it run on an MCU with MicroPython. It is actually to use Cython to convert Python code into C code so it can run on an MCU running MicroPython. The performance gains from doing this is what makes it worth it. I have done this many times when I want to have my Python code run faster. Compile it as a C extension module and presto 50 to 200 times faster. That ability would be fantastic to have with MicroPython especially if dealing with things like displays and custom rendering directly to a frame buffer. Doing those kinds of operations in Python code is painful to watch it run because of how slow it is. It's far worse on an MCU. Think Road Runner vs Wile E. Coyote as a comparison. Giving a user a way to get that speed without having to go through the very long process of learning C would be an awesome thing to be able to do.

dlech, 2025-02-23:
Have you tried the native emitter or viper emitter for better performance?

https://docs.micropython.org/en/latest/reference/speed_python.html

traverseda, 2025-02-23:
Wow, old post. The main reason I could see for doing this is writing a micropython library that is compatible with CPython. I guess what I was really asking for all those years ago was for someone to write native/viper emitters that are compatible with CPython, as vanilla python should be compatible with CPython anyway.

kdschlosser, 2025-02-24:
@dlech 

I am aware of both the native code emitter and also the viper code emitter. The native code emitter consumes very large amounts of RAM even when the code is not actually running. the viper code emitter is designed to deal with byte arrays and to do math as well as bit operations. It is not as heavy on the RAM use as the native code emitter is but it still consumes more RAM than C code or even Python code. I believe any interaction with Python objects outside of integers and byte arrays is slower than doing it from Python code.