micropython/micropython #5572: tests/basics/parser.py failes with python 3.8

dlech, 2020-01-24:
```
$ python3 --version
Python 3.7.6
$ python3 tests/basics/parser.py 
SyntaxError
SyntaxError
SyntaxError
```

```
$ python3 --version
Python 3.8.1
$ python3 tests/basics/parser.py 
SyntaxError
SyntaxError
SyntaxError
Traceback (most recent call last):
  File "./tests/basics/parser.py", line 30, in <module>
    compile("\\\n", "stdin", "exec")
  File "stdin", line 1
    \
    ^
SyntaxError: unexpected EOF while parsing
```

jimmo, 2020-02-03:
FWIW -- a .exp file was added in mid-December in 624f4ca39bbbf9c89fce24f4051dcdd9417e59b1 so it shouldn't be necessary to run `basics/parser.py` with CPython.

It would be good to understand why CPython's behavior changed here though.

dlech, 2020-02-03:
Thanks. I should have made sure to test with the latest version.

I looked through the CPython changelog, but nothing jumped out at me as possibly related.

closed 2020-02-03 by dlech