micropython/micropython #5577: ussl issue when using network.PPP over GSM ESP32

minusplusminus, 2020-01-24:
Hi, 

I'm using an ESP32 combined with SIM800. I'm porting my working code from LOBO micropython to this repo because LOBO lacks BLE. 

I've tested my code on the following firmware:

esp32spiram-idf3-20191220-v1.12.bin
esp32spiram-idf3-20200124-v1.12-86-gcb4472df4.bin

```

import machine
import sys
import time

try:
    import usocket as _socket
except:
    import _socket
try:
    import ussl as ssl
except:
    import ssl

DEBUG=True

def logline(str):
    if DEBUG:
        print(str)

def log(str):
    sys.stdout.write(str)

def connect():
    from machine import UART

    GSM_PWR = machine.Pin(4, machine.Pin.OUT)
    GSM_RST = machine.Pin(5, machine.Pin.OUT)
    GSM_MODEM_PWR = machine.Pin(23, machine.Pin.OUT)

    GSM_PWR.value(0)
    GSM_RST.value(1)
    GSM_MODEM_PWR.value(1)

    GSM = UART(1, baudrate=115200, timeout=1000, rx=26, tx=27)

    GSM_APN = '[APN]'
    logline('Waiting for AT command response...')
    atcmd('AT', GSM)
    atcmd('ATZ', GSM)
    time.sleep_ms(500)
    atcmd('ate0', GSM)
    atcmd('AT+CPIN?', GSM)
    atcmd('AT+CREG?', GSM)
    atcmd('AT+CNMI=0,0,0,0,0', GSM)

    logline("Connecting to GSM...")

    atcmd('AT+CGDCONT=1,"IP","' + GSM_APN + '"', GSM)
    atcmd('AT+CGDATA="PPP",1', GSM)

    return connectGPRS(GSM)

def atcmd(cmd, GSM, wait=500, tries=20):
    GSM.write(cmd + '\r\n')
    time.sleep_ms(wait)

    for retry in range(tries):
        resp = GSM.read()
        if resp:
            print(resp)
            return resp
        else:
            log('.')
            time.sleep_ms(wait)
    else:
        raise Exception("Modem not responding!")

def ssl_test(use_stream=True):
    s = _socket.socket()

    ai = _socket.getaddrinfo("google.com", 443)
    print("Address infos:", ai)
    addr = ai[0][-1]

    print("Connect address:", addr)
    s.connect(addr)

    s = ssl.wrap_socket(s)
    print(s)

    if use_stream:
        # Both CPython and MicroPython SSLSocket objects support read() and
        # write() methods.
        s.write(b"GET / HTTP/1.0\r\n\r\n")
        print(s.read(4096))
    else:
        # MicroPython SSLSocket objects implement only stream interface, not
        # socket interface
        s.send(b"GET / HTTP/1.0\r\n\r\n")
        print(s.recv(4096))

    s.close()

def connectGPRS(GSM):
    import network
    GPRS = network.PPP(GSM)
    GPRS.active(True)
    GPRS.connect()

    for retry in range(20):

        ret = GPRS.isconnected()

        if ret:
            break
        else:
            log('.')
            time.sleep_ms(2000)
    else:
        raise Exception("Modem not responding!")

    logline(GPRS.ifconfig())
    ssl_test()
    return GPRS
```

both of them return `MBEDTLS_ERR_SSL_CONN_EOF   -0x7280` as an error

thanks in advance!

minusplusminus, 2020-01-24:
oh wait, my code example works. I think it's related when data is requested and the import is done. Still unclear what the behaviour is.

minusplusminus, 2020-01-24:
can't reproduce it again

closed 2020-01-24 by minusplusminus

chtinnes, 2021-03-14:
Hey, I have pretty much the same issue. Do you know what it was?

chtinnes, 2021-03-15:
For me, it looks like this is one possible behavior when there is not enough memory left for the SSL cipher. 
Using another cipher with lower memory requirements did the job =)