micropython/micropython #5581: No tell() in BytesIO

dogtopus, 2020-01-25:
Attempting to use `tell()` in `BytesIO` results in an AttributeError.

robert-hh, 2020-01-25:
Instead you could use: 
a.seek(0,  1)

dogtopus, 2020-01-26:
> Instead you could use:
> `a.seek(0, 1)`
> Funny enough, there is a tell function in stream.c, but no QSTR for the API.

Yea that could work. However I don't really like a lot of workarounds in the code. Besides that it should be easy to just add it to MicroPython. It should have the pointer for the current position already so just a simple subtraction would probably do it.

peterhinch, 2020-01-26:
I don't think so: `seek()` returns no value.

robert-hh, 2020-01-26:
In my tests seek did (and the code is written that way). 

N.B.: I would not have posted that without trying.

peterhinch, 2020-01-26:
Sorry, you're right. After further digging this seems to be a behaviour which changed at some point in Python's past.

dpgeorge, 2020-03-11:
Since file objects have `tell()` it's nice to make `StringIO` and `BytesIO` to also have that method (to make them easier to use interchangeably).  It's also a very minor addition, so done in ed93778e004a2c6c2160f54d7362f14c5797d9c6

closed 2020-03-11 by dpgeorge