micropython/micropython #5586: ESP32: KeyboardInterrupt raised in thread while typed in UART REPL.

yannpom, 2020-01-28:
Hello,

Here is how to reproduce my problem:

- I start a task A in a new thread.
- Then I start another task B in the main thread (UART REPL).
- Then I press CTRL+C.

**Expected behaviour:**
    Task B gets a KeyboardInterrupt exception. Task A continues running.

**Actual behaviour:**
    KeyboardInterrupt is randomly raised in task A or B.

Here is the full log:
```
MicroPython v1.12-96-gc3095b37e-dirty on 2020-01-28; ESP32 module with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import _thread
=== import time
===
=== def long_process(name):
===     for i in range(1000):
===         print(name, i)
===         time.sleep(1)
===
=== _thread.start_new_thread(long_process, ("thread",))
===
=== long_process("main")
main 0
thread 0
main 1
thread 1
main 2
thread 2
main 3
thread 3
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "<stdin>", line 7, in long_process
KeyboardInterrupt:
>>> thread 4
thread 5
thread 6
thread 7

>>> thread 8
thread 9

>>>
MPY: soft reboot
MicroPython v1.12-96-gc3095b37e-dirty on 2020-01-28; ESP32 module with ESP32
Type "help()" for more information.
>>>
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import _thread
=== import time
===
=== def long_process(name):
===     for i in range(1000):
===         print(name, i)
===         time.sleep(1)
===
=== _thread.start_new_thread(long_process, ("thread",))
===
=== long_process("main")
main 0
thread 0
main 1
thread 1
main 2
thread 2
Unhandled exception in thread started by <function long_process at 0x3ffe4e20>
Traceback (most recent call last):
  File "<stdin>", line 7, in long_process
KeyboardInterrupt:
main 3
main 4
main 5
main 6
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "<stdin>", line 7, in long_process
  File "<stdin>", line 7, in long_process
KeyboardInterrupt:
>>>
```

During the first try you can see that after the KeyboardInterrupt the thread task is still running. This is the excepted behaviour.

Then after a soft reboot it's the opposite. The thread task got the interrupt and the main task is still running.

My understanding is that the UART received chars are handled in an interrupt, and when the CTRL+C char is detected it raise the KeyboardInterrupt exception in the current running thread.

Is this the normal behaviour or is this a bug?

Thanks,
Yann.

jefftharris, 2020-10-27:
We are experiencing a similar issue when reading from stdin.  A ctrl-c is sent to either the main or a background thread.  Should the state mp_pending_exception be thread-specific?  Is there a case for other exceptions to jump between threads based on which calls mp_handle_pending first?

agatti, 2026-05-23:
On current git `master` the problem reported here seems to be not occur, however there's something slightly more concerning happening.

If you run the reproduction code on an ESP32 via `mpremote run <file>`, CTRL+C to quit mpremote, and connect to the board's serial port you'll see the script is still running.

However, if you CTRL+C again from the serial port, the board performs a soft reboot due to the main thread exiting but without stopping the other thread:

```
$ mpremote.py run repro.py
main 0
thread 0
main 1
thread 1
^C
$ picocom -b 115200 /dev/ttyUSB0
picocom v3.1
[...]
Type [C-a] [C-h] to see available commands
Terminal ready
main 4
thread 4
main 5
thread 5
<CTRL+C pressed here>
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "<stdin>", line 7, in long_process
KeyboardInterrupt: 
>
MicroPython v1.29.0-preview.280.g8f3a416bb5.dirty on 2026-05-22; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> thread 6
thread 7
thread 8
```