micropython/micropython #5590: zlib compress

dadokkio, 2020-01-29:
Hi,
there is the possibility to compress with uzlib?

I've seen that there are some old issues closed with useful discussion but I wasn't able to find nothing usable.
Did someone make progress on this and could help? 
Thanks

jimmo, 2020-02-03:
@dadokkio  uzlib is enabled and supported on most ports, and documented here: http://docs.micropython.org/en/latest/library/uzlib.html

Could you provide more information about what you're looking for?

dadokkio, 2020-02-03:
I'm actually using that library but it supports only decompress. 
I'm trying to perform compression and like the library says:
"Compression is not yet implemented."

street-grease-coder, 2021-11-18:
assumed zlib has both compress and decompress functions, is it hard to port this?

Andronx, 2022-10-18:
Has something changed in the situation? I'm also looking for any micropython method to compress file.

jimmo, 2022-10-19:
The linked PR above has an implementation of this -- https://github.com/micropython/micropython/pull/5613

(Note this isn't merged yet so you'd have to rebase and build it from source)

There are kind of two things to consider here:
 - Should we add this to MicroPython but disabled-by-default. At least this makes it easy for people to enable for their own firmware.
 - Which boards should we enable it on?

@Andronx  I don't have a good feel for how useful zlib compression is on a microcontroller -- what sort of data are you compressing, is zlib a good fit? (My point is that the sort of data that a microcontroller might generate would often be something like time series samples from a sensor, which is not necessarily a good fit for zlib).

Andronx, 2022-10-19:
@jimmo Thank you for an answer. You asked good question. I'm using circuitpython on RP2040 board with 8MB flash and sdcard. I have few MB of txt log data, and I would like to acces these data from circuitpython usb storage. Plan is to copy that to from sd card to flash before start (in boot.py), that was the reason why I starded looking for compression. I expeced easy solution, but the more I  get into this topic I can't find easy solution. Ofcourse one of resons is that standard implementations of zip/deflate, operating on all input data stored in ram, on microcontrollers we need operate on small chunks. Time duration of such compression on is another topic. Anyway I see more people on different boards are interesed in some form of files compression.

jonnor, 2024-07-03:
The `deflate` module has support for both compression and decompression. So this issue can probably be closed now.

dpgeorge, 2024-07-04:
Compression in the `deflate` module is not enabled by default, but that's a separate issue (which feature level to enable compression on).

closed 2024-07-04 by dpgeorge