micropython/micropython #5600: BLE Server for ESP32

francescofact, 2020-02-01:
Hi. 
i just started with micropython. but i can't get a BLE server working on my ESP32 with micropython.
I want to connect to my esp32 via my phone with an app like "BLE IO".
I'm using ubluetooth and this is my code:

```
import bluetooth
bt = bluetooth.BLE()
bt.active(True)
bt.gap_advertise(1000)
```

but this goes into "OSError: [Errno 5] EIO". I can't understand how to do it.

dpgeorge, 2020-02-02:
This question is better suited to https://forum.micropython.org

In short: BLE has concepts of "central" and "peripheral", but not "server".  Examples of both are provided in `examples/bluetooth/`, `ble_temperature_central.py` and `ble_temperature.py`.

closed 2020-02-02 by dpgeorge

dhalbert, 2020-02-02:
@dpgeorge I am going to be pedantic, sorry. BLE does have client and server, which are orthogonal to central and peripheral. All four terms are "official". A server provides a service; a client talks to a server, accessing its characteristics. Once a client and a peripheral connect, either one can be a server to the other.

dpgeorge, 2020-02-03:
@dhalbert yes, of course, you are right!  I was being too simplistic and just thinking of the peripheral/central/observer/broadcaster roles, forgetting about things like GATT server/client.  Yes it's complex.