micropython/micropython #5615: False==0 and True==1 are no longer true expressions

dpgeorge, 2020-02-07:
A regression was introduced in 3aab54bf434e7f025a91ea05052f1bac439fad8c such that `False==0` and `True==1` now return `False` and not `True` as per CPython.

There are currently no tests for these.  Potential test is;
```python
print(False == 0)
print(0 == False)
print(True == 1)
print(1 == True)

s = set()
s.add(0)
s.add(False)
print(s)

s = set()
s.add(False)
s.add(0)
print(s)

s = set()
s.add(1)
s.add(True)
print(s)

s = set()
s.add(True)
s.add(1)
print(s)

s = {False, True, 2}
print(len(s))

d = {}
d[False] = 'false'
d[0] = 'zero'
print(d)

d = {}
d[0] = 'zero'
d[False] = 'false'
print(d)

d = {}
d[True] = 'true'
d[1] = 'one'
print(d)

d = {}
d[1] = 'one'
d[True] = 'true'
print(d)

d = {False:10, True:11, 2:12}
print(d[0], d[1], d[2])
```

dpgeorge, 2020-02-07:
The simplest fix might be to force the bool type to require full eq testing:
```diff
--- a/py/objbool.c
+++ b/py/objbool.c
@@ -86,6 +86,7 @@ STATIC mp_obj_t bool_binary_op(mp_binary_op_t op, mp_obj_t lhs_in, mp_obj_t rhs_
 
 const mp_obj_type_t mp_type_bool = {
     { &mp_type_type },
+    .flags = MP_TYPE_FLAG_NEEDS_FULL_EQ_TEST,
     .name = MP_QSTR_bool,
     .print = bool_print,
     .make_new = bool_make_new,
```
That will pass all above tests.  The only downside is that for tests of bool to bool (eg `False == True`) it's not very efficient because it must go through 3 nested calls to `mp_obj_equal_not_equal()` to convert the args to ints before it can compare them (it'll try `False == True` then `0 == True` then `0 == 1`, which will finally return false).

oliver-joos, 2020-02-07:
Thank you! I confirm that this patch fixes my test cases too.

I think the efficiency of "bool1 == bool2" is not important. [PEP8](https://legacy.python.org/dev/peps/pep-0008/#programming-recommendations) says "Don't compare boolean values to True or False using ==". So it is more readable and more efficient to replace "b == True" → "b" and "b == False" → "not b" and to compare two boolean expressions using "is" or "is not". No need for "bool1 == bool2".

dpgeorge, 2020-02-10:
> I think the efficiency of "bool1 == bool2" is not important.

The equality function is used also for searching objects in a dict/set, so if the dict/set has bools as keys then it'll be slower to look up.

> The only downside is that for tests of bool to bool (eg `False == True`) it's not very efficient because it must go through 3 nested calls to `mp_obj_equal_not_equal()` to convert the args to ints before it can compare them

But, actually, this was anyway the behaviour before this change to equality testing: for bool compared to int it would have 2 nested calls to `mp_obj_equal`, and for a bool-bool comparison when they were different (ie `False==True` or `True==False`) it would have 3 nested calls.

dpgeorge, 2020-02-10:
See #5628 for a fix.

PR #5628 (2020-02-11): py: Expand type-equality flags to 3 separate ones to fix bool/namedtuple equality
Fixes #5615 and #5620, with tests for both.
closed 2020-02-11 by dpgeorge