micropython/micropython #5620: namedtuple are no longer compareable with tuple

oliver-joos, 2020-02-07:
Another regression was introduced in 3aab54b such that namedtuples are not equal to the tuple they were created from:
```
>>> from collections import namedtuple
>>> MyTuple = namedtuple('MyTuple', ('a', 'b', 'c'))
>>> t = (4, 5, 6) ; mt = MyTuple(*t)
>>> mt == t                  # <--- is True in CPython 3 and MicroPython < 3aab54b
False
>>> isinstance(mt, tuple)    # although namedtuple is still a subclass of tuple
True
```

This bug is related to issue #5615.

oliver-joos, 2020-02-07:
It seems that this issue may be fixed with a patch similar to that in #5615.

**But please note:**
This patch is a blind guess without understanding the whole new equality testing!
```diff
--- micropython/py/objnamedtuple.c
+++ micropython_lisa/py/objnamedtuple.c
@@ -154,6 +154,7 @@
 
 STATIC mp_obj_t mp_obj_new_namedtuple_type(qstr name, size_t n_fields, mp_obj_t *fields) {
     mp_obj_namedtuple_type_t *o = mp_obj_new_namedtuple_base(n_fields, fields);
+    o->base.flags = MP_TYPE_FLAG_NEEDS_FULL_EQ_TEST,
     o->base.base.type = &mp_type_type;
     o->base.name = name;
     o->base.print = namedtuple_print;
```

dpgeorge, 2020-02-10:
Thanks for the report, what's most interesting (to me) is that there were no tests for this.

dpgeorge, 2020-02-10:
See #5628 for a fix.