micropython/micropython #5622: `with` statement in loop takes cyclically increasing time

dhalbert, 2020-02-08:
```python
import pyb

class WithTest:
    def __init__(self):
        pass
    def __enter__(self):
        pass
    def __exit__(self, *exc):
        pass

x1 = pyb.Pin(pyb.Pin.board.X1, pyb.Pin.OUT_PP)

# CYCLICALLY INCREASING INTERVALS
wt = WithTest()
while True:
    with wt as wtx:
        x1.value(not x1.value())

#OK: CONSTANT INTERVALS
while True:
    with WithTest() as wt:
        x1.value(not x1.value())
```
The first loop in this test program produces a cyclically increasing waveform. I tested on a PyBoard v1.1 with v1.94 and the also the latest build `pybv11-20200208-v1.12-154-gce40abcf2.dfu`

The period increases from about 38 microseconds to about 240 microseconds. Then there's a gap (gc?) and it starts over.

If you comment out the first loop and use the second (which constructs the `with` object over and over), it runs at a constant 180 microseconds or so.

The fact that it's a `with` statement is important; The `as` is not necessary. We first discovered this while doing I2C transactions in CircuitPython which were taking too long. It took a while to figure out that it was not an I2C issue, but was solely due to the `with`.

I don't see why the `with` should inherently cause this behavior, so I think there's something going on in the VM that's slowing things down on each iteration, probably allocating storage which finally gets gc'd. But why should we see an increasing interval? It's like something is walking down an increasingly longer list.

This is easiest to see on an oscilloscope:
https://youtu.be/tGB2uVwig_4  (unlisted; use the link)

Screen shots from Saleae:
first loop (taken at gap: starts fast after a gap and slows down):
[image]

second loop (constant before and after gap):
[image]

Also filed as https://github.com/adafruit/circuitpython/issues/2602.

stinos, 2020-02-08:
> it's like something is walking down an increasingly longer list.

Didn't check in detail what the culprit is here but that is indeed exactly how allocation works (and I assume using `with` i.e. setting up a context requires allocation) so each loop cycle there's an allocation and the next free memory block for the next allocation is further away from the start of the heap and takes longer to find. Adding a `gc.collect()` call in each iteration could 'fix' that but performace will be absymal. You can also use `micropython.mem_info()` to get basic information on what goes on regarding allocation.

dhalbert, 2020-02-08:
That makes sense. I thought it might be allocating some kind of context, but I didn't expect the time to allocate would vary so much. I haven't spent much time in that code. The allocator might keep track of the last location of blocks of a certain size or something like that, skip over completely full blocks, etc. (Maybe it already does some tricks.)

In the short term, we should avoid `with` in time-critical loops.

tagging @ladyada and @tannewt

stinos, 2020-02-08:
> Maybe it already does some tricks.

Some yes, but e.g. what you propose doesn't solve much: there's still allocation in a loop so at one point the heap is full and it will need to be gargage collected and you get a severe hit.

> In the short term, we should avoid with in time-critical loops.

Yes. Unless `with` can somehow be implemented without allocation or can yield the memory it used back immediately or until there's something like scoped allocation or so. So currently the principle is simple: if you need uSec timing and allocation takes some time in that range, there's not much else to do than not allocate.

dhalbert, 2020-02-08:
>~~In the short term, we should avoid `with` in time-critical loops.~~

I realize that's too simplistic an answer for the issue. The actual code is of course a lot more complicated than this: we read a value from a sensor, parse it, do something with the value, etc. all of which allocate their own storage. Those allocations probably swamp the `with` allocation. If allocation can be made faster, that's nice, though there will still be the occasional gc hiccup. In the long run, to reduce latency, we need incremental gc or reference counting. Neither is going to happen in the short run.

jepler, 2020-02-08:
This version seems to give good performance (note that *args are not used in `__exit__` but instead 3 named arguments are used):
```
class WithTest:
    def __init__(self):
        pass
    def __enter__(self):
        pass
    def __exit__(self, unused1, unused2, unused3):
        pass

wt = WithTest()
for _ in range(1048576):
    with wt as wtx:
        pass
```
I ran this on micropython unix port and compared the overall runtime.  Dan's "constant time" code and this code both run in under 1s, while the "increasing intervals" code takes over 6s.

Linux `perf` profiling shows that in the "increasing intervals" code, 97% of all time is taken in  this call stack (basically, in m_malloc):
```
    99.64%     0.12%  micropython  micropython         [.] mp_execute_bytecode
            |          
             --99.52%--mp_execute_bytecode
                       |          
                       |--97.05%--mp_call_method_n_kw
                       |          |          
                       |           --97.02%--mp_call_function_n_kw
                       |                     |          
                       |                      --96.98%--fun_bc_call
                       |                                |          
                       |                                 --96.74%--mp_setup_code_state
                       |                                           |          
                       |                                            --95.37%--mp_obj_new_tuple
                       |                                                      |          
                       |                                                       --95.36%--m_malloc
                       |                                                                 gc_alloc
```

stinos, 2020-02-08:
Good find, I mistakingly assumed it was the context itself allocating but admit I didn't see in the source code where that happened, then gave up. So it's the arguments, shame on me :)

tannewt, 2020-02-09:
MicroPython is a bit smarter with it's allocations. It keeps track of the next free block so it doesn't always start at the beginning. You can see it here: https://github.com/micropython/micropython/blob/master/py/gc.c#L474 with the `MP_STATE_MEM(gc_last_free_atb_index)`. It does appear that this only happens when the allocation is a single block though.

Super interesting that it is made worse by the `*args`.

peterhinch, 2020-02-09:
In one of his talks on optimisation @dpgeorge said that *args and **kwargs are relatively slow.

jimmo, 2020-02-09:
This is very similar to a few other related issues that have come up recently. See https://forum.micropython.org/viewtopic.php?f=20&t=7029&p=39960 for a nice variant (thanks to @robert-hh ). You can also make it happen with just a simple function call in a loop (where the function has enough locals that it allocates its frame on the heap).

There are some challenges to addressing this without making fragmentation or general performance worse. I looked into this a bit a few weeks ago and came up with some fixes... haven't tested this on your `with` example above, but it helps a lot with my other test cases (and improves the perf test scores, on STM32 at least) -- https://github.com/jimmo/micropython/commit/f7201e0f38217861e7e9a8e5e19ce36c3785555c
Note that it just makes the search faster, so the problem still exists just is far less noticeable. Very much WIP, needs some work and a lot of testing to get to PR. Also unsure how well it will work on architectures that don't have CTZ (or CLZ/RBIT).

dpgeorge, 2020-02-10:
> In one of his talks on optimisation @dpgeorge said that *args and **kwargs are relatively slow.

Yes, that's because they will allocate a tuple/dict respectively upon entry to the function.  So don't use them if performance/no-memory-allocation is important.