micropython/micropython #5624: Socket.listen(backlog) leads to OSError: 112

mrwhy-orig, 2020-02-08:
Currently I'm trying to listen to a socket. My setup is utilizing uasyncio. 

But the same error is happening with plain use of sockets as well. 

I'm building the master branch with EspIDF 3.3.1 

`
import usocket as _socket
ai = _socket.getaddrinfo("0.0.0.0", 80, 0, _socket.SOCK_STREAM)
ai = ai[0]
s = _socket.socket(ai[0], ai[1], ai[2])
s.setblocking(False)
s.setsockopt(_socket.SOL_SOCKET, _socket.SO_REUSEADDR, 1)
s.bind(ai[-1])
s.listen(10)
`

mrwhy-orig, 2020-02-09:
I tested the extmod-uasyncio from @dpgeorge that branch is working. Then I checked the differences. The difference is found is, that the branch isn't using this commit:
98a3911c430b0cc96e1821d7ca589b9be3355fc3 py/scheduler: Add "raise_exc" argument to mp_handle_pending.

I moved back before this commit and it is working again.

mrwhy-orig, 2020-02-09:
The only way I got around this error was to use the uasyncio branch from @dpgeorge. But this is a work in progress branch and unfortunately not 100% compatible with my code. There are other problems with the Stream class...

mrwhy-orig, 2020-02-10:
Using the IP address of the board in the getaddrinfo method, is working with latest µP release.

dpgeorge, 2020-02-11:
Running the above code twice does show the problem on the second run.  What's happening here is that the socket `s` is still active and listening (in the background) and it's not possible to have another socket listen on the same address (even using SO_REUSEADDR won't help).

Existing sockets must be cleaned up / closed before starting a new server on the same IP/port.

I've pushed commit 5a755ac30ad0636e17868e19e6d77b41d8be9ff5 to help provide a better error message in such a case.

mrwhy-orig, 2020-02-11:
Thanks for looking into this. Based on your explanation I searched my code for double port usage.

I'm creating 2 server instances one on port 80 and one on port 8081. That is working fine. The thing is/was I'm checking for internet connectivity in my ota class.

```
def testConnectivity(self):
    ai = usocket.getaddrinfo("http://b-have.de", 80, 0, usocket.SOCK_STREAM)
    if len(ai)> 0:
      return True
    else:
      return False
```

Is it possible that the getaddrinfo method doesn't release the port properly? I've seen that i tries to check with mdns.