micropython/micropython #5627: nrf24l01 driver pipes addresses compatibility with arduino RF24 driver

eagle1361, 2020-02-09:
I have been trying to get this nrf24l01 driver on an ESP32 to hear transmissions from arduino based systems running the popular RF24 library (https://github.com/nRF24/RF24). After setting everything to the same settings (and confirming by checking the registers), it didn't receive any data from the arduinos until I reversed the RX pipe address byte array. This driver and the RF24 library appear to write the 5 byte array in reverse orders. 

For example, if the arduino has the address defined as: 
const uint64_t pipes[2] = {0xF0F0F0F0E1LL, 0xF0F0F0F0D2LL};

This driver needs to have the addresses defined like this: 
pipes = (b'\xe1\xf0\xf0\xf0\xf0', b'\xd2\xf0\xf0\xf0\xf0')
not:
pipes = (b'\xf0\xf0\xf0\xf0\xe1', b'\xf0\xf0\xf0\xf0\xd2')

Not sure of the solution except to reverse the address. The differences may have to do with the nrf24l01+ expecting the addresses to be written with the LSB first or a difference in how the byte arrays are stored in memory.

peterhinch, 2020-02-10:
This is because the Arduino C code defines the numbers as 64 bit integers. Somewhere in the Arduino code these are being transmitted as sequences of bytes in "little endian" format i.e. least significant byte first.

The MicroPython driver represents these numbers as little endian bytes objects. This directly maps onto the sequence in which they are transmitted and is clearer in its intent.

This is not a bug. The little endian format is something which should be (perhaps is) documented at the Arduino end, in a guide to portability to other platforms.

eagle1361, 2020-02-10:
I understand this is not a bug but the differences between how the addresses are stored caused several days of struggles getting the code to work. Other than the addresses, all the other settings were straight forward to set for matching the Arduino settings. 

The fact that the Arduino code stores the address with the most significant byte "first" (when reading left to right) and in the MicroPython driver you need to format the address with the least significant byte at the left could be hard for the hobbyist programmer to figure out. I almost abandoned MicroPython on the ESP32 because of this, until I finally found the clues in the Arduino source code and the nrf24l01+ data sheet that byte order might be the issue. 

Perhaps a comment in the nrf24l01test.py example script would help users know the order expected for the addresses since the pipes addresses appear to be identical in both the Arduino C example code and the MicroPython example code, when reading from left to right. I don't know the proper github way to add a comment to the example code.

peterhinch, 2020-02-11:
Sorry, I thought you were reporting this as a bug in the driver.

I agree this is arguably a bug in the test program in that the choice of value seems to imply Arduino compatibility when in fact there is none. I'll raise a PR to insert a suitable comment.

dpgeorge, 2020-02-16:
Addressed by d2f22ea9531ee7e61d3e3c8b818797fd3e947b01

closed 2020-02-16 by dpgeorge

danielsalinas228, 2021-06-19:
Hi @eagle1361!
This post is kind of old but I´m trying to set up a communication between a RP Pico and an Arduino, I've already tested the communication between two Pico's but I'm struggling to do the same with the Arduino. Could u send me ur code on the Arduino Side?

eagle1361, 2021-06-19:
It's not hard...just make sure you setup all the settings the same. There
are some defaults in Arduino that can bite you. Also confirm that the
Arduino is talking to the chip. My code isn't for public release.

danielsalinas228, 2021-06-19:
@eagle1361 I think my issue is with the payload, I've setup the payload in the Pico as 4 but I'm not sure how to set this up in the Arduino. Basically I´m just sending a 1 and a 0 from the Pico to the Arduino:
#include <nRF24L01.h>
#include <RF24.h>
#include <RF24_config.h>
#include <SPI.h>

const int ledPin = 3;
int ledState = 0; 

RF24 radio(9,10);
const uint64_t pipe = 0xF0F0F0F0E1LL;
int msg[1];

void setup() {
  // put your setup code here, to run once:
  pinMode(ledPin,OUTPUT); 
  Serial.begin(9600);
  radio.begin();
  radio.openReadingPipe(1, pipe);
  radio.startListening();
}

void loop() {

  if(radio.available()){
    int done = radio.read(msg, 1);
    Serial.println(msg[0]);
  }
}