micropython/micropython #5629: esp32: Floating point arithmetics with native .mpy modules

carstenblt, 2020-02-10:
Anytime I want to do a division or comparison with floats, I get an undefined symbol error by the linker. How can I include them?

E.g. a division yields this error:
`LinkError: build/mylib.o: undefined symbol: __divsf3`

dpgeorge, 2020-02-10:
At the moment this is not supported because the linker doesn't know how to automatically provide these software-float routines.

As a workaround, look for an implementation of `__divsf3` for esp32 (in C or assembler) and include it in the build.

carstenblt, 2020-02-10:
I see, thank you. The esp32 has a FPU, but apparently they only included support for it in the latest toolchain according to their esp-idf v4 release notes...

mbooth101, 2022-07-31:
This catches me out too. It can be trivially reproduced with the following sample:

### float_division.c
```
#include "py/dynruntime.h"

STATIC mp_obj_t v_normalise(mp_obj_t factor) {
        mp_float_t f = mp_obj_get_float(factor);
        return mp_obj_new_float(f / (mp_float_t)0.3333);
}
STATIC MP_DEFINE_CONST_FUN_OBJ_1(v_normalise_obj, v_normalise);

mp_obj_t mpy_init(mp_obj_fun_bc_t *self, size_t n_args, size_t n_kw, mp_obj_t *args) {
        MP_DYNRUNTIME_INIT_ENTRY;
        mp_store_global(MP_QSTR_v_normalise, MP_OBJ_FROM_PTR(&v_normalise_obj));
        MP_DYNRUNTIME_INIT_EXIT;
}
```
### Makefile
```
MPY_DIR = ../micropython
MOD = float_division
SRC = float_division.c
ARCH = xtensawin
CLEAN_EXTRA = float_division.mpy
include $(MPY_DIR)/py/dynruntime.mk
```

With ESP-IDF 4.4 I see the following linker error: 

```
$ make V=1
/bin/mkdir -p build/
GEN build/float_division.config.h
python3 ../micropython/tools/mpy_ld.py '-vvv' --arch xtensawin --preprocess -o build/float_division.config.h float_division.c
CC float_division.c
xtensa-esp32-elf-gcc -I. -I../micropython -std=c99 -Os -Wall -Werror -DNDEBUG -DNO_QSTR -DMICROPY_ENABLE_DYNRUNTIME -DMP_CONFIGFILE='<build/float_division.config.h>' -fpic -fno-common -U _FORTIFY_SOURCE  -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT  -o build/float_division.o -c float_division.c
LINK build/float_division.o
python3 ../micropython/tools/mpy_ld.py '-vvv' --arch xtensawin --qstrs build/float_division.config.h -o build/float_division.native.mpy build/float_division.o
qstr vals: v_normalise
qstr objs: 
LinkError: build/float_division.o: undefined symbol: __divsf3
make: *** [../micropython/py/dynruntime.mk:145: build/float_division.native.mpy] Error 1
```

ma261065, 2023-05-22:
Is there any way to call the ROM FP functions, like the Ardunio Core module does?

For example, see https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/tools/sdk/esp32/ld/esp32.rom.libgcc.ld

jonnor, 2023-07-29:
Has anyone gotten a solution to this? 
There is a [forum thread](https://forum.micropython.org/viewtopic.php?t=11513) that says one can link libgcc.a into the module. But no details as to exactly how to do that in a native module.

jonnor, 2023-07-30:
I managed to find a solution, by extracting .o files from libgcc and adding them to the build.
I got my extension build working for xtensa, as well as armv6m and arm7m.
https://github.com/emlearn/emlearn-micropython/blob/078938a29c46ace4acfda388128e2f7ade5d6801/eml_trees/Makefile#L18

jonnor, 2023-07-31:
There is also a relevant discussion at https://github.com/orgs/micropython/discussions/11730

jimmo, 2023-08-03:
Updated the title as this affects all ports not just esp32.

hard-fp works in dynamic native modules, but I'm not sure if soft-fps will ever be supported as it's a lot of complexity for the dynamic loader (unless we're prepared to pay a lot of extra overhead in the .mpy file to duplicate all the soft-fp support, which I think is the solution that @jonnor is referring to above).

The rp2040 might be an interesting exception because although it uses soft-fp, the implementation is at a known location in mask rom (rather than being provided by the compiler). So in some sense from the perspective of this issue, it's more like hard-fp. It will require some special cases to implement though.

jonnor, 2024-05-05:
The overhead of soft floating point being duplicated in the module is not a dealbreaker in my opinion. The modules I have implemented so far are 2-10 kB in size. This is a very acceptable size for many devices and applications. Even right now, where these modules are loaded into RAM. When it would be possible to load these into FLASH (as I have heard some talks about), then it would be even better.
And supporting this does not bloat anything that does not actually make use of these features. So the choice on whether to utilize it would be completely on those that develop with MicroPython.

Note that even on hard-fp targets (such as ESP32/xtensawin and ARM Cortex M4F), occasionally some support functions from libgcc/libm are needed for some floating point operations. So the lack of integrated support for resolving such symbols is not limited to soft-fp devices (such as Cortex M0+).

niftyhacking, 2024-05-05:
On Sun, May 5, 2024 at 6:21 AM Jon Nordby ***@***.***> wrote:
>
> The overhead of soft floating point being duplicated in the module is not a dealbreaker in my opinion. The modules I have implemented so far are 2-10 kB in size.

Be specific about the floating point support you are asking for.
The short list is: 32 bit, 64 bit floats with IEEE full exception support.
There is a recommended arithmetic operations list.
.... exponents. logarithms  and trig functions.

https://en.wikipedia.org/wiki/IEEE_754

Since a 64 bit int has a lot of range most micropython should
facilitate 64bit integer operations and bit field ops on 32 and 64 bit
that facilitate IEEE float data import and export. Early microprocessor vendors
had fast floating point libraries before hardware FPU hardware was common.
There are C and C++ libraries in the public domain.  Minor adjustments
for endian
and also inline ASM when justified.

The OLD fast float strategies could be valuable for IOT applications.

-- 
    T o m    M i t c h e l l  (on NiftyEgg[.]com )

jonnor, 2024-05-05:
I would like the dynamic native modules to have the same level floating point support as user C modules. Though personally, I only care about single precision floating point.

jonnor, 2024-08-09:
Would an MR that adds support for automatically resolving symbols from GCC be welcomed? CC @jimmo  @dpgeorge 
Either as an integrated part of mpy_ldd.py or alternatively as an additional tool that bridges mpy_ldd.py and Makefile - I have started building some tools for this in emlearn-micropython (mentioned in previous comment), and I could improve it further to make it a proper drop-in solution. With examples on how to use in the natmod directory, built as part of CI pipeline.

jonnor, 2024-08-09:
Current tooling for finding symbols is here:
https://github.com/emlearn/emlearn-micropython/blob/master/tools/find_symbols.py
And the usage - which is still quite manual, is here:
https://github.com/emlearn/emlearn-micropython/blob/master/src/emliir/Makefile#L22
An upstreamed approach would have to streamline this, to limit/eliminate having to specify the symbols manually

sjaak31367, 2024-09-03:
Greetings!  
The tool for missing symbols really helped me (thank you!), though now the "undefined symbol" gets a bit weird, it seems that math.h is used during compiling, but not linking, as many (if not all) of the math functions throw the same error.  
boards: esp32 (Wemos D32 Lite) and esp32s2 (Lolin S2 Mini)  
error:
```
/home/sjaak/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/esp32/libgcc.a
LINK _extendsfdf2.o
LinkError: build/floatstuff.o: undefined symbol: sqrt
make: *** [/home/sjaak/Documents/micropython/micropython/py/dynruntime.mk:150: build/floatstuff.native.mpy] Error 1
```

files to reproduce:  
Makefile  
```makefile
# Location of top-level MicroPython directory
MPY_DIR = /home/sjaak/Documents/micropython/micropython

# Architecture to build for (x86, x64, armv7m, xtensa, xtensawin)
ARCH = xtensawin

# Name of module
MOD = floatstuff

# Source files (.c or .py)
SRC = floatstuff.c

# Stuff to make soft-float work
# If symbols are undefined, use tools/find_symbols.py to locate object files to add
SOFTFP_O := _extendsfdf2.o _truncdfsf2.o
SOFTFP_ENABLE=1
SRC_O += $(SOFTFP_O)

# With or without doesn't change a thing
#CFLAGS += -lm -Im

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk

# CROSS is defined by the included
LIBGCC_FILENAME = $(shell $(CROSS)gcc $(CFLAGS) -print-libgcc-file-name)
$(info $(LIBGCC_FILENAME))

_extendsfdf2.o:
	$(CROSS)ar -x $(LIBGCC_FILENAME) $(SOFTFP_O)
_truncdfsf2.o:
	$(CROSS)ar -x $(LIBGCC_FILENAME) $(SOFTFP_O)
```

floatstuff.c  
```c
#include "py/dynruntime.h"
#include <math.h>
static mp_obj_t py_floatstuff(mp_obj_t f) {
    return mp_obj_new_float_from_f(floor(cos(sqrt(mp_obj_get_float(f)))));
}
static MP_DEFINE_CONST_FUN_OBJ_1(py_floatstuff_obj, py_floatstuff);
mp_obj_t mpy_init(mp_obj_fun_bc_t *self, size_t n_args, size_t n_kw, mp_obj_t *args) {
    MP_DYNRUNTIME_INIT_ENTRY
    mp_store_global(MP_QSTR_floatstuff, MP_OBJ_FROM_PTR(&py_floatstuff_obj));
    MP_DYNRUNTIME_INIT_EXIT
}
```
Removing math.h, and adding definitions in floatstuff.c does work... But only on my esp32, not the esp32s2. Maybe my esp32s2 has a faulty FPU or something, or maybe worth a second eye?

vshymanskyy, 2024-09-13:
@carstenblt please check out this PR: https://github.com/micropython/micropython/pull/15838

dpgeorge, 2025-03-17:
This is now working, see above PR.

closed 2025-03-17 by dpgeorge