micropython/micropython #5630: nRF9160 and Micropython

ijustwant, 2020-02-10:
Hi
I'm compiling .hex files(secureboot.hex and firmware.hex) according to this link and info at this git. But i cannot get the REPL on any serialport. Am i missing something? I have succsessfully written the files to the 10090/nRF9160 device.  Please advise.

glennrub, 2020-02-10:
The issue is because the firmware on the nRF52 that configures the switches/wires to the Segger chip takes a bit to boot. The 52 is used as a board controller to switch lines to be used internally on the board, or routed out on the pin lists. It's up to the firmware to set this up. The default firmware configures the lines for UART from the nRF91. I measured the UART to be done with configuration to be roughly 150ms - 200ms. Hence, the Segger chip does not detect flow control being used at bootup as the lines are floating.

There are two ways around this:
1) Disable UART flow control in boards/pca10090/mpconfigboard.h
2) My prefered. Ground the RTS and toggle a Pin Reset on the board. Personally i have soldered a wire from Test Point 7 (Labeled TP7) which i always ground.

ijustwant, 2020-02-11:
Ok thank you.
But i deleted the firmware in the nRF52. Is this still applicable ?

glennrub, 2020-02-11:
If you are interested in the Button/Switches and LED's on the board, you might need it. For UART's I'm pretty sure.

You can download `Board Controller Firmware` from this page: https://www.nordicsemi.com/Software-and-Tools/Development-Kits/nRF9160-DK/Download#infotabs. The name `nRF9160_DK_board_controller_FW.hex ` might sound a bit misleading. However, this is the hex file that Nordic's documentation points to.

glennrub, 2020-02-11:
@ijustwant, you are right! You do not need any firmware on the nrf52. https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf91_dk%2FUG%2Fnrf91_DK%2Fboard_controller.html. The default routing should be good enough for a micropython REPL.

I also tested erasing my firmware on the nrf52 and i get a REPL, with working LED's. You can still try to keep the TP7 over a button reset, and see if that helps.

glennrub, 2020-02-11:
Looks like it only works for me if the "PROG/DEBUG" switch is in nrf91 state. Maybe that's a hint.

ijustwant, 2020-02-12:
The option one above worked: Disable UART flow control :-)
Thasnk you very much! I really appreciate your help :)

ijustwant, 2020-02-12:
Hi again :)
Could you do me a favour and compile an hex file for this code:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/nrf9160/serial_lte_modem/README.html

I don't have any setup to build myself, but i need this hex :)

jonnor, 2023-07-30:
It seems that a solution was found for the reported issue.

Unfortunately, it is unlikely that others can build software for a specific user or usage. One should set up ones own build environment in order to develop with MicroPython.

closed 2023-07-31 by mattytrentini