micropython/micropython #5650: mem_info reports impossible values

osresearch, 2020-02-15:
Tracing `m_malloc()` and `m_free()` calls, it looks like the free rarely gets called, so the reported memory usage keeps climbing and `current_bytes_allocated` is significantly larger (34KB) than the heap + stack (29KB) after a while:

```
>>> micropython.mem_info()
mem: total=43806, current=33642, peak=34475
stack: 428 out of 5512
GC: total: 24000, used: 7888, free: 16112
 No. of 1-blocks: 184, 2-blocks: 26, max blk sz: 18, max free sz: 373
```

This might be a bug in my port; I'm using the minimal `gc_collect()` that has a comment about it "_may function incorrectly_".

dpgeorge, 2020-02-16:
> it looks like the free rarely gets called, so the reported memory usage keeps climbing and `current_bytes_allocated` is significantly larger (34KB) than the heap + stack (29KB) after a while:

Yes that is correct.  Since most heap memory is freed via the GC these accounting numbers are not balanced, in terms of alloc+free.

I'd suggest to just disable the `MICROPY_MEM_STATS` option, and instead use the accurate values from the "GC: total: ..." line.

osresearch, 2020-02-16:
Thanks for the insight; that matches what I had deduced from the tracing.  Should the `MICROPY_MEM_STATS` option be removed or flagged as deprecated since the numbers are not accurate?

dpgeorge, 2020-02-16:
> Should the `MICROPY_MEM_STATS` option be removed or flagged as deprecated since the numbers are not accurate?

It could still be useful to get some stats on start-up memory use (eg how much the parser/compiler use, which do free most of their memory correctly).  Also if another heap allocator was added (as an alternative to gc.c) this could also be useful.

So if there's to be any action here I would suggest adding a comment in `py/mpconfig.h` above the `MICROPY_MEM_STATS` option stating its limitations.