micropython/micropython #5652: ESP8266 - Issues with COM port disappearing in Windows 10 when connected to wifi

joewez, 2020-02-16:
I have recently upgraded from 1.10 to 1.12 and have noticed a new issue I am having in Windows 10 with my LOLIN D1 Mini devices (USB interface is CH340).  Whenever I connect to a wifi network, the COM port suddenly disappears and my putty session is broken.  In the Device Manager, the COM port continues to fluctuate, indicating that the ESP8266 is doing something....

I have tried turning ESP debug messages on and off, but it does not make a difference.

It only seems to occur if wifi is connected to a network.

I can duplicate the issue in the nightly build.

dpgeorge, 2020-02-18:
It sounds like a hardware issue, possibly combined with wifi power saving.  Can you test v1.11 to try and narrow down the problem?

joewez, 2020-02-27:
I have tested v1.11 and cannot duplicate the issue,  Let me re-check the currently nightly.

joewez, 2020-03-27:
OK. I have narrowed it down to a hardware/power issue.  Closing ticket.

closed 2020-03-27 by joewez