micropython/micropython #5655: Compilation crash when no modules are frozen

Bockwuhrst, 2020-02-17:
If manifest.py is empty I get:
```
error freezing mpy []: b'usage: mpy-tool.py [-h] [-d] [-f] [--merge] [-q QSTR_HEADER]\n
[-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]\n
[-o OUTPUT]\n
files [files ...]\nmpy-tool.py: error: the following arguments are required: files\n'
../../py/mkrules.mk:103: recipe for target 'build-PYBV11/frozen_content.c' failed
make: *** [build-PYBV11/frozen_content.c] Error 1
```

This is running `make BOARD=PYBV11` on Windows WSL

If I add `freeze('$(MPY_DIR)/drivers/onewire', 'onewire.py')` to manifest.py compilation works fine.

dpgeorge, 2020-02-18:
Thanks for the report.  I've proposed a fix for this in #5656

dpgeorge, 2020-02-21:
Fixed by a636837987f7d53b1965b02c15050075f8ad4152

closed 2020-02-21 by dpgeorge

Bockwuhrst, 2020-02-21:
Awesome. Thanks!