micropython/micropython #5658: Meet Error while config MICROPY_USE_READLINE = 0 of unix port

swortain, 2020-02-18:
Hi, I want to use UART as REPL port on Raspberry Pi.
At first I try to config stdin input, so I change "MICROPY_USE_READLINE = 1" to "MICROPY_USE_READLINE = 0" in mpconfigport.mk. But the error comes out when I run make on raspberry Pi as this

```
...
...
...
CC modffi.c
CC ../../lib/timeutils/timeutils.c
LINK micropython
/usr/bin/ld: build-standard/py/modbuiltins.o: in function `mp_builtin_input':
modbuiltins.c:(.text.mp_builtin_input+0x30): undefined reference to `readline'
collect2: error: ld returned 1 exit status
make: *** [../../py/mkrules.mk:146：micropython] Error 1
```

dlech, 2020-02-18:
Disabling MICROPY_USE_READLINE will make the REPL less useful.

On RaspberryPi, you can use `raspi-config` to enable a command shell on the serial port then just log in via the serial port and run `micropython` from the command line.

swortain, 2020-02-19:
> Disabling MICROPY_USE_READLINE will make the REPL less useful.
> 
> On RaspberryPi, you can use `raspi-config` to enable a command shell on the serial port then just log in via the serial port and run `micropython` from the command line.

👌，I‘ll try it and thx for the reple

dpgeorge, 2020-02-19:
Thanks for the report.  Should be fixed by 410757f4f4d0ba6a6e2a80d3ef7f86317339c54c

closed 2020-02-19 by dpgeorge