micropython/micropython #5666: Document existence of brew installable micropython for Mac

george-hawkins, 2020-02-19:
I've just started using MicroPython with a ESP32 board - it's absolutely super! Now I want to work in a MicroPython environment without having to connect to the board. It turns out all I have to do to install such an environment on my Mac is:

    $ brew install micropython
    $ micropython
    MicroPython v1.12 on 2020-01-11; darwin version
    Use Ctrl-D to exit, Ctrl-E for paste mode
    >>> 

However, I only came across this by chance. If I look at the ["Unix version" section](https://github.com/micropython/micropython#the-unix-version) of the main `README` or the [Mac section](https://github.com/micropython/micropython/wiki/Getting-Started#mac-osx) of the "Getting Started" wiki page then I just see fairly involved instructions for building from sources.

I tried `brew install micropython` on the off chance that it existed after seeing that packages have been built for other platforms, e.g. Fedora.

The [micropython formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/micropython.rb) is up-to-date, i.e. currently provides MicroPythin 1.12, and seems to have been actively maintained since April 2016 when the original formula for 1.7 was added.

Micropython is amazing and anything that makes it easier to use on any platform is of great value. I know Micropython is a huge effort undertaken by a limited number of people, so my apologies for submitting this as an issue, rather than a pull request.

george-hawkins, 2020-02-19:
And it might be worth pointing out that the UNIX port has no `help` function. One of the first things I tried on starting `micropython` on my Mac was:

```
>>> help("modules")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'help' isn't defined
```

My first assumption was that there was a bug in the brew formula and it had failed to properly install anything beyond the interpreter itself (I expected next to find that it didn't know about any modules, but it does).

It turns out there is no `help` builtin in the UNIX port - see commit b8cf948. I guess the reason for this is as commented in #450, from way back in 2014, where @dpgeorge states:

> The unix/ port is not a priority, since it's main aim is to test the uPy core. Hence why no help() has been implemented for it.

Although if I look at how `help` is implemented in [`py/builtinhelp.c`](https://github.com/micropython/micropython/blob/master/py/builtinhelp.c) I can't really see a good reason why this default help shouldn't be enabled for the UNIX port? Being able to see 
what modules are available really is helpful when getting started.

dlech, 2020-02-19:
builtin help is already enabled by default in the unix port. It is controlled by the `MICROPY_PY_BUILTINS_HELP` option (and `MICROPY_PY_BUILTINS_HELP_MODULES`).

dpgeorge, 2020-02-20:
It might be worth adding "brew install micropython" to the unix quickref doc, at the top.

> builtin help is already enabled by default in the unix port

It's enabled only on the coverage build.  See #1354 for a past discussion about adding help to the standard unix build.  Since it's an often-requested thing, probably worth enabling.