micropython/micropython #5669: Descriptor attribute access on a class doesn't call __get__

tannewt, 2020-02-19:
CPython calls `__get__` with `obj` as `None` when a descriptor is accessed on the class itself rather than an instance. MicroPython returns the descriptor instance instead of calling `__get__`.

dpgeorge, 2020-02-19:
Thanks for the report.  The following code shows the issue:
```python
class A:
    def __get__(self, obj, cls):
        print('__get__', self, obj, cls)
        return 123
        
class B:
    x = A()
    
print(B.x)
print(B().x)
```
Output for CPython:
```
__get__ <__main__.A object at 0x7f4de09d2a00> None <class '__main__.B'>
123
__get__ <__main__.A object at 0x7f4de09d2a00> <__main__.B object at 0x7f4de0982610> <class '__main__.B'>
123
```
Output for MicroPython:
```
<A object at 7f189ac5e280>
__get__ <A object at 7f189ac5e280> <B object at 7f189ac5e260> <class 'B'>
123
```
At the very least this should be added to the list of differences to CPython.

tannewt, 2020-02-20:
Thanks for the repro! I'll follow up here if I hunt it down.