micropython/micropython #5673: 499 VS 1661? Why performance reduced a lot when use the newest source build for esp32 port?

ccccmagicboy, 2020-02-20:
Please help!
On the same hardware board and same cpu freq setting, also use upip.install("micropython-pystone_lowmem") to get the same test code, got results below:

```shell
I (20) pm_esp32: Frequency switching config: CPU_MAX: 240, APB_MAX: 240, APB_MIN: 240, Light sleep: DISABLED
MicroPython cddb90e-dirty on 2020-02-18; ESP32 module with ESP32
Type "help()" for more information.
>>>[reboot detected 0]repl is in normal command mode
[\r\x03\x03] b'\r\n>>> '
[\r\x01] b'\r\n>>> \r\nraw REPL; CTRL-B to exit\r\n>' Ready.

import machine
machine.freq(240*1000*1000)
print(machine.freq())

import network
sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect('***', '***')

import upip
upip.install("micropython-pystone_lowmem")

import pystone_lowmem
pystone_lowmem.main()

Pystone(1.2) time for 500 passes = 1001ms
This machine benchmarks at 499 pystones/second

```

but, after load a earlier version fw from official download page, got the result below:

```shell
MicroPython v1.12 on 2019-12-20; ESP32 module with ESP32
Type "help()" for more information.
>>>[reboot detected 0]repl is in normal command mode
[\r\x03\x03] b'\r\n>>> '
[\r\x01] b'\r\n>>> \r\nraw REPL; CTRL-B to exit\r\n>' 

import machine
machine.freq(240*1000*1000)
print(machine.freq())

import network
sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect('***', '***')

import upip
upip.install("micropython-pystone_lowmem")

import pystone_lowmem
pystone_lowmem.main()

Pystone(1.2) time for 500 passes = 301ms
This machine benchmarks at 1661 pystones/second

```

some one can help?

jimmo, 2020-02-20:
Please see https://github.com/micropython/micropython/issues/5040#issuecomment-524334492 for a possible explanation, but I agree that is quite dramatic. Is this a 3.3 or 4.0 build? With or without SPIRAM?

jimmo, 2020-02-20:
FWIW, on a TinyPico, I see:
master: BOARD=TINYPICO: 1385 pystones/second
master: BOARD=GENERIC: 1170 pystones/second
v1.12: BOARD=TINYPICO: 1736 pystones/second
v1.12: BOARD=GENERIC: 735 pystones/second

The first are built from source, the second two are from the downloads page.

ccccmagicboy, 2020-02-20:
@jimmo source code build with IDF3 and use board GENERIC, and another fw from download page esp32-idf3-20191220-v1.12.bin

ccccmagicboy, 2020-02-20:
test with source code which download page v1.12 used(esp32-idf3-20191220-v1.12.bin, commit id: 1f371947309c5ea6023b6d9065415697cbc75578), the result even more lower: 
```
Pystone(1.2) time for 500 passes = 1403ms
This machine benchmarks at 356 pystones/second
```
maybe it bear on my frozen modules???