micropython/micropython #5674: Memoryview comparison not working correctly anymore

kevinkk525, 2020-02-20:
In an older release this worked:

```
>>> memoryview(b"hi")==b"hi"
True
```

Now it doesn't:

```
>>> memoryview(b"hi")==b"hi"
False
```

This causes all kinds of problems in my code, obviously.
My guess is that this is caused by the commit about equality comparison (can't find it right now but read it today before pulling the latest changes)

peterhinch, 2020-02-20:
Nasty. It works in CPython. A search on **is:pr equality** throws up several possible candidates.

dpgeorge, 2020-02-20:
Thanks for the report.  IMO it's good to have all these issues exposed, it means the test suite was not testing things which were important and used in real code.

commit 3ccce89b (2020-02-21): py/objarray: Turn on MP_TYPE_FLAG_EQ_CHECKS_OTHER_TYPE for memoryview.

And add corresponding tests.

Fixes #5674 (comparison of memoryview against bytes).
closed 2020-02-21 by dpgeorge