micropython/micropython #5677: Truth value test of class deriving from native class can raise TypeError

stinos, 2020-02-20:
Don't know if this has always been like this, but it looks like a bug and otherwise it should probably be mentioned as a difference with CPython for instance. Python standard says
```
By default, an object is considered true unless its class defines either a
__bool__() method that returns False or a __len__() method that returns zero,
when called with the object.
```
But when taking a native type which doesn't implement `__bool__` nor `__len__` this happens:
```
import uctypes

class C:
  pass

class B(uctypes.struct):
  def __init__(self):
    super().__init__(1, 'i')

def TestBool(x):
  print(type(x), x, 'compares', True if x else False)

TestBool(C())
TestBool(uctypes.struct(1, 'i'))
TestBool(B())
```

output:

```
<class 'C'> <C object at 3a7f20> compares True
<class 'struct'> <struct ERROR 1> compares True
Traceback (most recent call last):
  File "D:\foo.py", line 43, in <module>
  File "D:\foo.py", line 39, in TestBool
TypeError: unsupported type for __bool__: 'struct'
```

So even though `struct` itself doesn't explicitly implement `__bool__`, MicroPython falls back to default truth testing. But for a derived class it does not. I cannot pinpoint the exact reason, but for the derived class the unary_op call is forwarded by `instance_unary_op` to `mp_unary_op` which raises an exception, whereas in the other cases (class C above and bare uctypes.struct), `mp_obj_is_true` just falls through. So maybe the logic used in `instance_unary_op` is not correct, or `mp_obj_is_true` could try to catch exceptions but that seems far fetched.

The workaround is to manually add something like
```
def __bool__(self):
  return not not super()
```

jimmo, 2020-02-20:
The code that implements the "by default an object is true" logic that you quoted is mp_obj_is_true (which is invoked directly via the `True if x else False`).

What happens here is that B's `.unary_op` (i.e. `instance_unary_op`) realises that the Python type doesn't have a `__bool__`, but it is derived from a native type, so it forwards to `mp_unary_op`. This works for many built-in types that have a `.unary_op` (e.g. list, set, str, etc) because they all have to implement their "not empty" `MP_UNARY_OP_BOOL` behavior. But for struct, which has no `.unary_op`, it then fails saying the operator isn't supported (never getting a chance to call mp_obj_is_true).

I _think_ the fix is that `mp_unary_op` should also special-case `MP_UNARY_OP_BOOL` like it does for `MP_UNARY_OP_NOT` to always call `mp_obj_is_true`.   (Alternatively, if the type has no `.unary_up` then it could at that point check if the operator is `MP_UNARY_OP_BOOL`, but `mp_obj_is_true` does the right thing with calling `.unary_op` anyway).

dpgeorge, 2020-02-21:
@stinos I assume you hit this with a custom native type, and `uctypes` is used here just to show the problem?

stinos, 2020-02-21:
@dpgeorge yes that's correct. Just needed to find a type without bool/len and this was the first one I came across

closed 2021-07-23 by dpgeorge

stinos, 2021-08-23:
Bit late, but can confirm this works correctly, thanks!

commit 4e39ff22 (2021-07-23): py/runtime: Fix bool unary op for subclasses of native types.

Previously a subclass of a type that didn't implement unary_op, or didn't
handle MP_UNARY_OP_BOOL, would raise TypeError on bool conversion.

Fixes #5677.