micropython/micropython #5679: ports/unix/mpconfigport.mk does not allow configuration/inclusion of FRAMEBUF

ikelos, 2020-02-20:
I was recently trying to test some micropython on my laptop that's targeted against an ESP8266.  I used the `framebuf` module, but found it wasn't included in my default installation of micropython.  I found that it was enabled in the "coverage" build, but I didn't want it significantly different from the standard build, I just wanted the `framebuf` module to be present.  So I looked for configuration options and found `mpconfigport.mk` and decided to add `MICROPY_PY_FRAMEBUF = 1` to it, but that didn't seem to change anything.  Compiling it with `V=1` showed that the Makefile wasn't passing it as a compilation flag, so whilst make knew it was defined, the C code did not.

It seems like have all configuration options being configurable in the same place (where a distribution can just add a line to the `.mk` file, rather than writing a define into the middle of a `.h` file) would make sense?  As best I can tell, the Makefile needs to look for the variable, and if present add `-DMICROPY_PY_FRAMEBUF=1`.  It doesn't feel like a big fix, but please let me know if you need more information or don't have time to handle it and I can try and mock up a pull request...

dpgeorge, 2020-02-21:
> So I looked for configuration options and found `mpconfigport.mk` and decided to add `MICROPY_PY_FRAMEBUF = 1` to it, but that didn't seem to change anything.

The vast majority of options need to be set in the `mpconfigport.h` file.  Only options that the linker must know about are configured via `mpconfigport.mk`, and such options are then passed through to the C compiler (preprocessor) via `-D<opt>=1`.

> It seems like have all configuration options being configurable in the same place (where a distribution can just add a line to the `.mk` file, rather than writing a define into the middle of a `.h` file) would make sense?

To change all options to be set via the mk file is not really an option at this stage (because there are hundreds of them).

But it should be easy enough to configure a specific build yourself.  Three ways of doing it are:
1. Use `make CFLAGS_EXTRA="-DMICROPY_PY_FRAMEBUF=1"`
1. Make a custom .h file and use `make CFLAGS_EXTRA="-DMP_CONFIGFILE=<myconfig.h>"`
1. Make a custom unix variant and use `make VARIANT=myvariant`

For (2) the custom `myconfig.h` can be as simple as:
```c
#define MICROPY_PY_FRAMEBUF (1)
#include "mpconfigport.h"
```

ikelos, 2020-02-21:
Ah, ok, my bad.  I misunderstood the intention of the `.mk` file, thanks for the alternative methods for getting the compiler flag through, I think the custom variant option looks the best.  5:)

closed 2020-02-21 by ikelos