micropython/micropython #5683: Failed to build

Chrithon, 2020-02-21:
I had tried many times, but it still fails
Results:
ying@ying-MS-7B22:~/esp/micropython-esp32/ports/esp32$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
16:45:18.132237 git.c:344               trace: built-in: git show -s '--pretty=format:%H'
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/ying/esp/esp-idf
Current git hash:   a0644bf8ae6a9435ed2ce46d0828e5970e5f4623
Supported git hash: 2c95a77cf93781f296883d5dbafcdc18e4389656
16:45:18.148349 git.c:344               trace: built-in: git describe
CC /home/ying/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c
/home/ying/esp/esp-idf/components/bootloader_support/src/bootloader_flash.c:16:10: fatal error: bootloader_flash.h: No such file or directory
 #include <bootloader_flash.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Makefile:897: recipe for target 'build/bootloader//home/ying/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o' failed
make: *** [build/bootloader//home/ying/esp/esp-idf/components/bootloader_support/src/bootloader_flash.o] Error 1

I followed the README file, but it still doesn't work.
（See issue: https://github.com/micropython/micropython-esp32/issues/1 ）

Chrithon, 2020-02-21:
Device:ESP32

robert-hh, 2020-02-21:
```
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path: /home/ying/esp/esp-idf
Current git hash: a0644bf8ae6a9435ed2ce46d0828e5970e5f4623
Supported git hash: 2c95a77cf93781f296883d5dbafcdc18e4389656
```
This is at least one of the reasons. To cure it, 
- go into the esp-idf directory and run the commands
git checkout 2c95a77cf93781f296883d5dbafcdc18e4389656
git submodule sync
git submodule update --init

- Then go into the esp32 port directory and run:
git submodule sync
git submodule update --init
make clean
make

And yes, the hint of Issue #1 should also be observed.

jimmo, 2020-02-21:
It also looks like you're using a much older version of MicroPython.

Edit: Actually you're trying to build micropython-esp32 -- please see https://github.com/micropython/micropython-esp32/#a-note-about-this-esp32-repository

Chrithon, 2020-02-21:
OK, I'll try.
(Don't close the issue!)

Chrithon, 2020-02-21:
I have waited for 5 minutes but it still doesn't work.
Results:
22:07:22.776185 pkt-line.c:80           packet:          git< 0000
* Couldn't find host github.com in the .netrc file; using defaults
* Found bundle for host github.com: 0x558615583dc0 [can pipeline]
* Re-using existing connection! (#0) with host github.com
* Connected to github.com (192.30.253.113) port 443 (#0)
> POST /espressif/esp32-bt-lib.git/git-upload-pack HTTP/1.1
Host: github.com
User-Agent: git/2.17.1
Accept-Encoding: gzip
Content-Type: application/x-git-upload-pack-request
Accept: application/x-git-upload-pack-result
Content-Length: 527

* upload completely sent off: 527 out of 527 bytes
< HTTP/1.1 200 OK
< Server: GitHub Babel 2.0
< Content-Type: application/x-git-upload-pack-result
< Transfer-Encoding: chunked
< Expires: Fri, 01 Jan 1980 00:00:00 GMT
< Pragma: no-cache
< Cache-Control: no-cache, max-age=0, must-revalidate
< Vary: Accept-Encoding
< X-GitHub-Request-Id: 79DF:0D4C:39804B:60328C:5E4FE41A
< X-Frame-Options: DENY
< 
22:07:23.638918 pkt-line.c:80           packet:   fetch-pack< NAK
22:07:23.638953 pkt-line.c:80           packet:   fetch-pack> 0000
22:07:23.639595 pkt-line.c:80           packet:     sideband< PACK ...
22:07:26.906234 run-command.c:646       trace: run_command: git index-pack --stdin --fix-thin '--keep=fetch-pack 21349 on ying-MS-7B22' --check-self-contained-and-connected --pack_header=2,739
22:07:26.909545 git.c:344               trace: built-in: git index-pack --stdin --fix-thin '--keep=fetch-pack 21349 on ying-MS-7B22' --check-self-contained-and-connected --pack_header=2,739
(It's still stuck there)
(Few minutes later......)
22:18:35.340432 pkt-line.c:80           packet:   fetch-pack< NAK
22:18:35.340471 pkt-line.c:80           packet:   fetch-pack> 0000
22:18:36.933418 pkt-line.c:80           packet:     sideband< PACK ...
22:18:38.282299 run-command.c:646       trace: run_command: git index-pack --stdin --fix-thin '--keep=fetch-pack 22101 on ying-MS-7B22' --check-self-contained-and-connected --pack_header=2,7144
22:18:38.285518 git.c:344               trace: built-in: git index-pack --stdin --fix-thin '--keep=fetch-pack 22101 on ying-MS-7B22' --check-self-contained-and-connected --pack_header=2,7144
So I will try tomrrow,. If it's sloved. I will close the issue.

Chrithon, 2020-02-22:
Thanks, I will try BLE instead of Bluetooth.

closed 2020-02-22 by Chrithon